ARMmbed/mbed-os #3190: "mbed export -m K64F -i eclipse_gcc_arm" fails on mbed-os 5.2

TeroJaasko, 2016-11-03:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: 
- Priority: Major 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**

**mbed-cli version:** 
0.9.10

**meed-os sha:**
e2617cc (HEAD, tag: mbed_lib_rev128, tag: mbed-os-5.2.1, tag: latest, origin/mbed-os-5.2)

**DAPLink version:**

**Expected behavior**
"mbed export -m K64F -i eclipse_gcc_arm" command produces functional project file just as it did previously.

**Actual behavior**
Export fails completely.

**Steps to reproduce**
```
/tmp/mbed-os-example-blinky$ mbed export -m K64F -i eclipse_gcc_arm
/usr/local/lib/python2.7/dist-packages/fuzzywuzzy/fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_STORAGE
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_NANOSTACK
Scan: FEATURE_THREAD_BORDER_ROUTER
Traceback (most recent call last):
  File "/tmp/mbed-os-example-blinky/mbed-os/tools/project.py", line 244, in <module>
    main()
  File "/tmp/mbed-os-example-blinky/mbed-os/tools/project.py", line 240, in main
    zip_proj=zip_proj, build_profile=profile)
  File "/tmp/mbed-os-example-blinky/mbed-os/tools/project.py", line 93, in export
    build_profile=build_profile, silent=silent)
  File "/tmp/mbed-os-example-blinky/mbed-os/tools/project_api.py", line 229, in export_project
    macros=macros)
  File "/tmp/mbed-os-example-blinky/mbed-os/tools/project_api.py", line 90, in generate_project_files
    exporter.generate()
  File "/tmp/mbed-os-example-blinky/mbed-os/tools/export/cdt/__init__.py", line 14, in generate
    super(Eclipse, self).generate()
  File "/tmp/mbed-os-example-blinky/mbed-os/tools/export/makefile/__init__.py", line 103, in generate
    raise NotSupportedException("This make tool is in development")
tools.utils.NotSupportedException: This make tool is in development
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /tmp/mbed-os-example-blinky/mbed-os/tools/project.py -i eclipse_gcc_arm -m K64F --source ." in "/tmp/mbed-os-example-blinky"
---
```

----------------------------------------------------------------
## Enhancement
Git bisect reveals that the commit 29125229308bf356e0bcd419489e9bd6e5e87216 has broken the export. "git revert 29125229308bf356e0bcd419489e9bd6e5e87216" on mbed-os fixes the problem and the project file can be produced again.

**Reason to enhance or problem with existing solution**

**Suggested enhancement**
Fix and/or add CI job to detect regressions.

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2016-11-03:
Thanks for reporting, I could reproduce the problem

cc @theotherjimmy

0xc0170, 2016-11-03:
This lines are to blame https://github.com/ARMmbed/mbed-os/blame/master/tools/export/makefile/__init__.py (for loop and else):

```

        for templatefile in \
            ['makefile/%s_%s.tmpl' % (self.NAME.lower(),
                                      self.target.lower())] + \
            ['makefile/%s_%s.tmpl' % (self.NAME.lower(),
                                      label.lower()) for label
             in self.toolchain.target.extra_labels] +\
            ['makefile/%s.tmpl' % self.NAME.lower()]:
            try:
                self.gen_file(templatefile, ctx, 'Makefile')
                break
            except TemplateNotFound:
                pass
        else:
            raise NotSupportedException("This make tool is in development")
```

No template found , thus if goes to else.

jankii01, 2016-11-03:
Wonder how difficult it would be to have a CI check for something like this?-)

sarahmarshy, 2016-11-03:
@theotherjimmy has an open PR for this. On mobile or I'd find it.

0xc0170, 2016-11-03:
@sarahmarshy I am going through the PR just now, thanks for the pointer - https://github.com/ARMmbed/mbed-os/pull/3187

0xc0170, 2016-11-03:
This will be fixed for 5.2.2

bridadan, 2016-11-03:
CI checks are in the works, keep an eye out :smile:

theotherjimmy, 2016-12-09:
@TeroJaasko #3187 is now merged, did this resolve your issue?

theotherjimmy, 2016-12-20:
Alright, I'm going to close this now. If it's still a problem, please re-open this issue.

closed 2016-12-20 by theotherjimmy

TeroJaasko, 2017-01-12:
Yes, the export works now. Thanks.