ARMmbed/mbed-os #3192: Include path issue? Local configuration problem after mbed5.1 to 5.2

EduardPon, 2016-11-03:
After migrating from 5.1 to 5.2.1, I have some compiling issues with include paths. For example the following compiler error is reported:
`.\features\TARGET_WIRED_IPV6\sal-nanostack-driver-k64f-eth\source\k64f_eth_nanostack_port.c:22:25: fatal error: arm_hal_phy.h: No such file or directory #include "arm_hal_phy.h"`

The arm_hal_phy.h file is present and located in \yy-mbed5\mbed-os\features\nanostack\FEATURE_NANOSTACK\sal-stack-nanostack\nanostack\platform as it should be.

Do I miss some essential json configuration? Any hints? Thanks in advance.

Build output: 
```
C:\_ccm\EduardPon\yy-mbed5>mbed-cli compile -j0 -t GCC_ARM -m xx_K64F_IPV6 -v
[mbed] Working path "C:\_ccm\EduardPon\yy-mbed5" (program)
[mbed] Exec "python -u C:\_ccm\EduardPon\yy-mbed5\mbed-os\tools\make.py -t GCC_ARM -m xx_K64F_IPV6 --source . --build .\.build\
xx_K64F_IPV6\GCC_ARM -v -j0" in C:\_ccm\EduardPon\yy-mbed5
Building project yy-mbed5 (xx_K64F_IPV6, GCC_ARM)
Scan: .
Scan: FEATURE_WIRED_IPV6
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_IPV4
Scan: FEATURE_IPV6
Scan: FEATURE_STORAGE
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
[DEBUG] Macros: -DFEATURE_LWIP=1 -D__MBED__=1 -DTARGET_xx -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4
_M7 -DMBED_BUILD_TIMESTAMP=1478169014.13 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HA
RDWARE_ALT -DTARGET_KPSDK_MCUS -DTOOLCHAIN_GCC -DDEVICE_STDIO_MESSAGES=1 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_A
NALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DTARGET_FF_ARDUINO -DDEVICE_SERIAL=1 -DTARGET_xx_K64F_IPV6 -DDEVIC
E_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_I2CSLAVE=1 -D__CORTEX_M4 -D__MBED_CMSIS_RTOS_CM -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBE
D -DFEATURE_COMMON_PAL=1 -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DFEATURE_STORAGE=1 -DTARGET_WIRED_IPV6 -DDEVICE_SLEEP=1 -DTOO
LCHAIN_GCC_ARM -DTARGET_FRDM -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DFEATURE_CLIENT=1 -DD
EVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4
Compile [  4.6%]: k64f_eth_nanostack_port.c
[DEBUG] Compile: C:\yotta\gcc\bin\arm-none-eabi-gcc -std=gnu99 -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-l
ength=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointe
r -Os -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DFEATURE_LWIP=1 -D__MBED__=1 -DTARGET_xx -D__FPU_PRESENT=1 -DTAR
GET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DMBED_BUILD_TIMESTAMP=1478169014.13 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFS
L_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DTOOLCHAIN_GCC -DDEVICE_STDIO_MESSAGES=1 -DTARGET_CORTEX_M -DT
ARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DTARGET_FF_ARDUINO -DDEVICE_
SERIAL=1 -DTARGET_xx_K64F_IPV6 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_I2CSLAVE=1 -D__CORTEX_M4 -D__MBED_C
MSIS_RTOS_CM -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBED -DFEATURE_COMMON_PAL=1 -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DFEATURE_S
TORAGE=1 -DTARGET_WIRED_IPV6 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_FRDM -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_
ANALOGIN=1 -DDEVICE_PWMOUT=1 -DFEATURE_CLIENT=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4 @.\.build\xx_K64F_IPV6\GCC_ARM\.i
ncludes_b9707cc4f97d30683993bea8c955e3c3.txt -include .\.build\xx_K64F_IPV6\GCC_ARM\mbed_config.h -MD -MF .\.build\xx_K
64F_IPV6\GCC_ARM\features\TARGET_WIRED_IPV6\sal-nanostack-driver-k64f-eth\source\k64f_eth_nanostack_port.d -o .\.build\xx_K64F_IPV6\
GCC_ARM\features\TARGET_WIRED_IPV6\sal-nanostack-driver-k64f-eth\source\k64f_eth_nanostack_port.o .\features\TARGET_WIRED_IPV6\sal-nanostack-driv
er-k64f-eth\source\k64f_eth_nanostack_port.c
[DEBUG] Return: 1
[DEBUG] Output: .\features\TARGET_WIRED_IPV6\sal-nanostack-driver-k64f-eth\source\k64f_eth_nanostack_port.c:22:25: fatal error: arm_hal_phy.h: No
 such file or directory
[DEBUG] Output:  #include "arm_hal_phy.h"
[DEBUG] Output:                          ^
[DEBUG] Output: compilation terminated.
Traceback (most recent call last):
  File "C:\_ccm\EduardPon\yy-mbed5\mbed-os\tools\make.py", line 293, in <module>
    toolchain))
  File "C:\_ccm\EduardPon\yy-mbed5\mbed-os\tools\build_api.py", line 452, in build_project
    resources.inc_dirs)
  File "C:\_ccm\EduardPon\yy-mbed5\mbed-os\tools\toolchains\__init__.py", line 807, in compile_sources
    return self.compile_queue(queue, objects)
  File "C:\_ccm\EduardPon\yy-mbed5\mbed-os\tools\toolchains\__init__.py", line 870, in compile_queue
    raise ToolException(err)
ToolException: .\features\TARGET_WIRED_IPV6\sal-nanostack-driver-k64f-eth\source\k64f_eth_nanostack_port.c:22:25: fatal error: arm_hal_phy.h: No
such file or directory
 #include "arm_hal_phy.h"
```

0xc0170, 2016-11-03:
cc @SeppoTakalo

SeppoTakalo, 2016-11-03:
The include file `arm_hal_phy.h` is under `FEATURE_NANOSTACK` have you defined that?

Within mbed OS 5.1 and 5.2 the `FEATURE_IPV6` was renamed to `FEATURE_NANOSTACK`

But if you need wired IPv6 you are suppose to use LwIP.
That linked in driver only works with Nanostack and only in border router. It does not provide API for Socket layers.

EduardPon, 2016-11-03:
Hello,
Ok, I have renamed the IPV6 to NANOSTACK in the target.json file and fixed the include issue:
`        "features": ["NANOSTACK"],`

Yes, we need also the wired and IPv6 support. In the future we want to support both, wired and wireless.  So LwIP is the alternative?

SeppoTakalo, 2016-11-03:
LwIP supports both IPv4 and IPv6.
Currently only through Ethernet (also some Wifi devices emulate Ethernet).

Nanostack is used when 6LoWPAN is required over IEEE 802.15.4 radio. 6LoWPAN is compressed IPv6, also used in Thread.

On border routers, 6LoWPAN is translated to IPv6 and router to Ethernet. This is the only device type that we support when running IPv6 over Ethernet with Nanostack.

So if your device is not going to be a border router, you should look for LwIP based solution.

I'll close this now, as the original issue is resolved. Feel free to ask more question in this thread if you need more information.
Thank you.

closed 2016-11-03 by SeppoTakalo