ARMmbed/mbed-os #3196: Add Support for STM DISCOVERY 769NI in mbed Compiler

sandeepach, 2016-11-04:
### Description
- Type: Bug 
- Related issue: `#2182`
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
DIS_769I

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.2
**mbed-cli version:** 
129

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**
mbed Online compiler should have supported DIS_769 board from ST

**Actual behavior**
Although mbed library has support for DIS_769NI, mbed Online compiler doesnt support DIS_769 board from ST

**Steps to reproduce**
Open mbed online Compiler. List of platforms doesnt have DIS_769I

bcostm, 2016-11-04:
This board is planned to be added in January 2017 the latest.

sg-, 2017-02-03:
@bcostm Has this been released?

bcostm, 2017-02-06:
Yes it has been released last week or the week before. I just checked and the compilation is ok for this board on the mbed IDE.

closed 2017-02-06 by 0xc0170