ARMmbed/mbed-os #3197: Compiling for mbed-dev broken

andcor02, 2016-11-04:
There seems to be an issue here when compiling for mbed-dev in the online-IDE for specific platforms. 

E.g. FRDM-K64F will not compile and Nucleo F401RE (Not tested much more).

LPC1768 seems to work

What is the issue here, its irrespective what is included in the main program.

Regards,

Andrea

adbridge, 2016-11-04:
@andcor02 
The latest release of mbed-dev included a restructuring of the folders containing targets, cmsis, platform and hal. In the release notes it should state this and that consequently some paths to include files may have changed. Any applications you thus have that include full, explicit paths may need to be looked at. 
I compiled blinky on numerous platforms including K64F with the latest mbed-dev with no problems.
Please check your include paths and update if necessary. If that still doesn't help then please provide details of the problem you are seeing including the compiler output and we may be able to take a look.

andcor02, 2016-11-07:
Hi :),

Followed your advice seems to working great now 

Thanks

closed 2016-11-07 by andcor02