ARMmbed/mbed-os #3199: mbed5.1->5.2 missing entropy_hardware_poll.c ?

EduardPon, 2016-11-04:
Got triggered by a linker error about the fact an entropy function could not be linked. 

For K64F specific, in the path `mbed-os\targets\TARGET_Freescale\TARGET_KSDK2_MCUS\TARGET_MCU_K64F`
the file entropy_hardware_poll.c seems not to be present anymore. Thanks in advance.

adbridge, 2016-11-04:
@EduardPon I will investigate

adbridge, 2016-11-04:
@EduardPon It looks like this PR 
https://github.com/ARMmbed/mbed-os/commit/132a5bbff3edabca7371d41f7676149a60186a5d
explains what happened .

EduardPon, 2016-11-04:
I have added the [RNG] feature but still the same issue. 
The macro DEVICE_TRNG is still not defined to compile the trng_api code. 

See our target definition copy below:

"K64F": {
  "supported_form_factors": ["ARDUINO"],
  "core": "Cortex-M4F",
  "supported_toolchains": ["ARM", "GCC_ARM", "IAR"],
  "extra_labels": ["Freescale", "KSDK2_MCUS", "FRDM", "KPSDK_MCUS", "KPSDK_CODE", "MCU_K64F"],
  "is_disk_virtual": true,
  "macros": ["CPU_MK64FN1M0VMD12", "FSL_RTOS_MBED", "MBEDTLS_ENTROPY_HARDWARE_ALT"],
  "inherits": ["Target"],
  "progen": {"target": "frdm-k64f"},
  "detect_code": ["0240"],
  "device_has": ["ANALOGIN", "ANALOGOUT", "ERROR_RED", "I2C", "I2CSLAVE", "INTERRUPTIN", "LOWPOWERTIMER", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SERIAL_FC", "SLEEP", "SPI", "SPISLAVE", "STDIO_MESSAGES", "STORAGE", "RNG"],
  "features": ["IPV4", "STORAGE"],
  "release_versions": ["2", "5"]
},
"OURPLATFORM_K64F_IPV6": {
  "supported_form_factors": ["ARDUINO"],
  "core": "Cortex-M4F",
  "default_toolchain": "GCC_ARM",
  "extra_labels": ["Freescale", "KSDK2_MCUS", "FRDM", "KPSDK_MCUS", "KPSDK_CODE", "MCU_K64F", "K64F", "WIRED_IPV6" ],
  "is_disk_virtual": true,
  "inherits": ["K64F"],
  "progen": {"target": "frdm-k64f"},
  "detect_code": ["0240"],
  "features": ["NANOSTACK", "NANOSTACK_FULL"],
  "release_versions": ["2", "5"],
  "device_name": "MK64FN1M0xxx12"
}

```
#if defined(DEVICE_TRNG)

#include <stdlib.h>
#include "cmsis.h"
#include "fsl_common.h"
#include "fsl_clock.h"
#include "trng_api.h"

void trng_init(trng_t *obj)
{
    (void)obj;
    CLOCK_EnableClock(kCLOCK_Rnga0);
    CLOCK_DisableClock(kCLOCK_Rnga0);
    CLOCK_EnableClock(kCLOCK_Rnga0);
}
```

adbridge, 2016-11-04:
@0xc0170

EduardPon, 2016-11-04:
I see "RNG" must be changed to "TRNG"? With this change, it builds. But I get quite some warnings:

[Warning] platform_mbed.h@21,1: "MBEDTLS_ENTROPY_HARDWARE_ALT" redefined
Compile [ 35.2%]: utest_types.cpp
Compile [ 35.5%]: aesni.c
[Warning] platform_mbed.h@21,1: "MBEDTLS_ENTROPY_HARDWARE_ALT" redefined
Compile [ 35.8%]: arc4.c
[Warning] platform_mbed.h@21,1: "MBEDTLS_ENTROPY_HARDWARE_ALT" redefined
Compile [ 36.1%]: asn1parse.c
[Warning] platform_mbed.h@21,1: "MBEDTLS_ENTROPY_HARDWARE_ALT" redefined
Compile [ 36.4%]: asn1write.c
[Warning] platform_mbed.h@21,1: "MBEDTLS_ENTROPY_HARDWARE_ALT" redefined

0xc0170, 2016-11-07:
@EduardPon 
I would like to get some more details for this. The release notes should contain the information needed for the update from 5.1 to 5.2. If they fail, we need to make certain the next one release won't.

Where was ``MBEDTLS_ENTROPY_HARDWARE_ALT`` ? I recall in targets previously, it was cleaned-up.

EduardPon, 2016-11-07:
It was in previously in:
`mbed-os\hal\targets\hal\TARGET_Freescale\TARGET_KSDK2_MCUS\TARGET_K64
`
Please look also in my other issues. There are some that have useful info for the migration note.

0xc0170, 2016-11-07:
> It was in previously in:
mbed-os\hal\targets\hal\TARGET_Freescale\TARGET_KSDK2_MCUS\TARGET_K64

Have you managed to remove those redefinition warnings?

EduardPon, 2016-11-08:
Yes,

Adding #ifndef:

plaform_mbed.h
```
#ifndef MBEDTLS_ENTROPY_HARDWARE_ALT

#if defined(DEVICE_TRNG)
#define MBEDTLS_ENTROPY_HARDWARE_ALT
#endif

#endif
```

0xc0170, 2016-11-08:
@andresag01 Can you have a look ? Is it anything missing or should be documented, or all is correct and this can be closed?

sg-, 2017-02-03:
@EduardPon @andresag01 Has this been resolved?

EduardPon, 2017-02-06:
No change, still on the mbed 5.2. Coming April we have planned to migrate to the latest released mbed version.

andresag01, 2017-02-06:
The issues related to mbed TLS discussed in the first part of this conversation seem to have been fixed.

closed 2017-02-07 by sg-