ARMmbed/mbed-os #3201: wait_us() does not delay long enough

infinnovation, 2016-11-04:
### Description
- Type: Bug
- Priority: ?

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**meed-os sha:**
e2617cc (mbed-os-5.2.1)

**Expected behavior**
wait_us(n) waits for at least n microseconds

**Actual behavior**
wait_us(2950) waits for as little as 1375 microseconds in tests, maybe less

**Steps to reproduce**

Essentially the following, maybe triggered by button press, whatever:
```C
uint32_t start = us_ticker_read();
wait_us(2950);
uint32_t taken = us_ticker_read() - start;
printf("%lu\r\n", taken);
```
----------------------------------------------------------------
## Analysis

If you read the wait_us source code you will see the bogus line which should be removed.  The fact that the shortfall can be more than 1ms is probably due to Thread::wait (and all the layers underneath it) rounding down to a number of ticks.

geky, 2016-11-04:
Correct me if I'm wrong, but it actually looks like this line is at fault:
https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_wait_api_rtos.cpp#L40

The value of `start` is taken from before that wait call, so the value of `us` shouldn't be changed. Put up a quick patch: https://github.com/ARMmbed/mbed-os/pull/3202

geky, 2016-11-09:
I believe this issue has been resolved by #3202. Feel free to reopen if I'm wrong.

closed 2016-11-09 by geky