ARMmbed/mbed-os #3203: Bug Thread.Start(object, Method) broken in mbed-os 5.1  & mbed-os 5,2

Roietronics, 2016-11-04:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug 
- Related issue: mbed-os Thread class
- Priority: Blocker 
---------------------------------------------------------------
## Bug
Thread.Start(object, Method) broken in mbed-os 5.1  & mbed-os 5,2
**Target**
??

**Toolchain:*
online

**Toolchain version:**

**mbed-cli version:** 
(`mbed -- 5.1 & 5.2`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**
The following code which would build under mbed 2 would still work on mbed 5
Class MyClass
{
public:
void MyFunction(void* args)
{
.
.
.
}
Thread MyThread();
osStatus status = MyTheard,start(this, MyFunction)
**Actual behavior**
Fatal compiler build error 
See mbed forum post for more details
https://developer.mbed.org/forum/bugs-suggestions/topic/27062/

**Steps to reproduce**
Attempt to build the sample demo program in the forum post 
----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

sarahmarshy, 2016-11-04:
I was able to use a similar function call to Thread::start (with 'this') as the object, and I was able to compile successfully. I did, however, see the generated warning. Wrapping the arguments in callback() solved the warning and compiled successfully.

Can you please paste all of the code where you are seeing this issue? I am having a difficult time reproducing your use case.

Roietronics, 2016-11-05:
The code is posted on the mbed  bug forum https://developer.mbed.org/forum/bugs-suggestions/topic/27062/).
The initial posting had been mangled by the forum's editor; however I have not successfully managed to make the code be un-mangled in the post please see the post

Roietronics, 2016-11-08:
Code is now viewable on mbed bug forum
https://developer.mbed.org/forum/bugs-suggestions/topic/27062/

sarahmarshy, 2016-11-08:
@Roietronics, I have updated the post with another code sample. The callback class is not thoroughly documented yet, so the confusion is understandable. Hopefully the new examples unblocks you.

closed 2017-02-03 by sg-