ARMmbed/mbed-os #3209: mbed_lib.json conflicts when compiling for a binary application

mottigondabi, 2016-11-06:
### Description
- Type: Bug
- Priority: Major

mbed CLI tool copies ***.json**, ***.a** files and potentially some other stuff that is given in `--source <path>`

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM|ARM

**Toolchain version:**
4.9.3

**mbed-cli version:** 
0.9.5

**meed-os sha:**
56f60a0

**Steps to reproduce**
1. Compile mbed-os as a static library (using --build libmbed-os)
2. Compile mbed-client as a static library (using --build libmbed-client)
3. Compile some `foo` application that has a single main.cpp (prints hello world or something) and uses `--source libmbed-os` and `--source libmbed-client` when compiling, e.g.:

`mbed compile -j1 --build .build-foo -t GCC_ARM -m K64F --artifact-name foo -v --source .build-libmbed-os --source .build-libmbed-client -v`

**Expected behavior**
`foo` application should cleanly compile

**Actual behavior**
`ConfigException: Library name 'mbed-os' is not unique (defined in '.build-libmbed-os/mbed_lib.json' and '.build-libmbed-client/mbed_lib.json')`

@jenia81 
@screamerbg 
@bridadan

bridadan, 2016-11-07:
@mottigondabi So to be clear, this is what your project looks like from a shared parent directory:

```
/mbed-os            # cloned from https://github.com/ARMmbed/mbed-os
/mbed-client        # cloned from https://github.com/ARMmbed/mbed-client
/foo                # your own project
```

Then you do the following steps:
1. `cd mbed-os`
2. `mbed compile --library --build .libmbed-os ...`
3. `cd ../mbed-client`
4. `mbed compile --library --build .libmbed-client ...`
5. `cd ../foo`
6. `mbed compile --source ../mbed-os/.libmbed-os --source ../mbed-client/.libmbed-client --source .`

And this gives you the following Exception:
`ConfigException: Library name 'mbed-os' is not unique (defined in '.build-libmbed-os/mbed_lib.json' and '.build-libmbed-client/mbed_lib.json')`

Is this correct?

theotherjimmy, 2016-11-07:
I take issue with the expected bevhavior: you are including the source of mbed and it's compiled form, so you should get an error like that.

0xc0170, 2016-11-16:
@mottigondabi Is this still an issue?

bridadan, 2017-01-17:
@mottigondabi bump #2, any update?

mottigondabi, 2017-01-22:
No. Issue is closed.

closed 2017-01-22 by mottigondabi

bqam-ublox, 2017-02-17:
This is usually due to another folder copied inside mbed os which contains a library with the same name as in mbed os due to which there is a conflict. Delete the already created library and it would build again. Its better to build it outside mbed os to avoid such issues using --build = <path>