ARMmbed/mbed-os #3210: [mbed-os rtos]: set_stack_heap() calculates wrong heap size when heap size is low

JojoS62, 2016-11-06:
### Description
- Type: Bug 

---------------------------------------------------------------
## Bug

**Target**
all

**Toolchain:**
all

**meed-os sha:**
actual master

In set_stack_heap(void) the mbed_heap start/size is calculated:

```    
    uint32_t free_size = HEAP_SIZE;
...
free_size -= mbed_stack_isr_size;
#endif

    /* Heap - everything else */
    mbed_heap_size = free_size;
    mbed_heap_start = free_start;
```

so when the HEAP_SIZE is low (due to many static vars in .bss) and less than mbed_stack_isr_size, the free_size becomes negative and the uint overflows. That gives you terrific 4GB of memory :-)

0xc0170, 2016-11-09:
@JojoS62 Thanks for reporting. Can you please send a patch for this bug? 

The freesize variable should not wrap around.

cc @c1728p9

JojoS62, 2016-11-09:
Solution sound easy, the check is simple, free_size must be larger than isr_stack_size. But what should be done when this test fails? 

How about this:
```
void set_stack_heap(void) {

    unsigned char *free_start = HEAP_START;
    uint32_t free_size = HEAP_SIZE;

#ifdef ISR_STACK_START
    /* Interrupt stack explicitly specified */
    mbed_stack_isr_size = ISR_STACK_SIZE;
    mbed_stack_isr_start = ISR_STACK_START;
#else
    /* Interrupt stack -  reserve space at the end of the free block 		*/
    /* if the heap size is too low, it's a severe fault				 		*/
    /* this means not enough RAM for rtos or too many variables in data+bss */
    if (mbed_stack_isr_size > free_size)
    	while(1) {};
    mbed_stack_isr_size = ISR_STACK_SIZE;
    mbed_stack_isr_start = free_start + free_size - mbed_stack_isr_size;
    free_size -= mbed_stack_isr_size;
#endif

    /* Heap - everything else */
    mbed_heap_size = free_size;
    mbed_heap_start = free_start;
}
```

I don't know what is the best error handling at this stage.

It should be a compile time check, but a simple
```
#if (ISR_STACK_SIZE > HEAP_SIZE)
    #error
#endif
```
does not compile, the preprocessor can not evaluate this expression.

geky, 2016-11-09:
Does `MBED_STATIC_ASSERT` from mbed_assert.h work?
``` cpp
MBED_STATIC_ASSERT(ISR_STACK_SIZE > HEAP_SIZE,
    "Not enough space reserved for the ISR stack");
```

JojoS62, 2016-11-09:
that gives the following error:
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:484:24: error: expression in static assertion is not constant

But the MBED_ASSERT compiles without errors:
```
void set_stack_heap(void) {

    unsigned char *free_start = HEAP_START;
    uint32_t free_size = HEAP_SIZE;

#ifdef ISR_STACK_START
    /* Interrupt stack explicitly specified */
    mbed_stack_isr_size = ISR_STACK_SIZE;
    mbed_stack_isr_start = ISR_STACK_START;
#else
    /* Interrupt stack -  reserve space at the end of the free block 		*/
    /* if the heap size is too low, it's a severe fault				 		*/
    /* this means not enough RAM for rtos or too many variables in data+bss */
    MBED_ASSERT(HEAP_SIZE >= ISR_STACK_SIZE);

    mbed_stack_isr_size = ISR_STACK_SIZE;
    mbed_stack_isr_start = free_start + free_size - mbed_stack_isr_size;
    free_size -= mbed_stack_isr_size;
#endif
```

sg-, 2017-02-03:
@JojoS62 @c1728p9 Has this been resolved?

PR #3898 (2017-03-09): Prevent underflow in heap size calculation
If the free memory on a device is small enough then the calculation to determine heap size could underflow to a large value. If this happens then malloc will never return failure and instead will cause a crash on allocation. This patch prevents the underflow so malloc works as expected even with low amounts of free memory.

This resolves #3210
closed 2017-03-09 by 0xc0170