ARMmbed/mbed-os #3212: Cannot debug anymore in mbed 5.2.

EduardPon, 2016-11-07:
I cannot debug anymore after switching to mbed 5.2.1. 
The option -o debug-info seems not to be supported anymore.
How can I enable it again? Thanks.

bulislaw, 2016-11-07:
We renamed this option to `--profile PROFILE     Build profile to use. Can be either path to jsonfile
                        or one of the default one (debug, default, small)` unfortunately there's a bug in mbed-cli (https://github.com/ARMmbed/mbed-cli/issues/370) displaying out of date help text which makes things harder to see.
So right now you need to use `--profile debug`

EduardPon, 2016-11-07:
How should we use this? Like this it does not work:

`mbed-cli compile -j0 -t GCC_ARM -m OUR_TARGET_K64F_IPV6 -o debug-info --profile PROFILE`

bulislaw, 2016-11-07:
Try `mbed-cli compile -j0 -t GCC_ARM -m OUR_TARGET_K64F_IPV6 --profile debug` or just `mbed compile ...`

bulislaw, 2016-11-07:
If you're not using recent version of mbed-os, you'll need to put whole path to the profile file eg `--profile mbed-os/tools/profiles/debug.json`

EduardPon, 2016-11-07:
I'm using the recent mbed-os, but the --profile debug returns the [ERROR] [Errno 13] Permission denied: 'debug'.

Adding in debug.json seems also not to work, see 2nd row:
```
    "GCC_ARM": {
        "common": ["-c", "-Wall", "-Wextra", "--profile", 
                   "-Wno-unused-parameter", "-Wno-missing-field-initializers",
                   "-fmessage-length=0", "-fno-exceptions", "-fno-builtin",
                   "-ffunction-sections", "-fdata-sections", "-funsigned-char",
                   "-MMD", "-fno-delete-null-pointer-checks",
                   "-fomit-frame-pointer", "-O0", "-g"],
        "asm": ["-x", "assembler-with-cpp"],
        "c": ["-std=gnu99"],
        "cxx": ["-std=gnu++98", "-fno-rtti", "-Wvla"],
        "ld": ["-Wl,--gc-sections", "-Wl,--wrap,main", "-Wl,--wrap,_malloc_r",
               "-Wl,--wrap,_free_r", "-Wl,--wrap,_realloc_r",
               "-Wl,--wrap,_calloc_r", "-Wl,--wrap,exit", "-Wl,--wrap,atexit"]
    },
```

This is what the debugger start reports in the console window:
```
No symbol table is loaded.  Use the "file" command.

Temporary breakpoint 1, 0x000448e4 in main ()
No breakpoint number 4.
```

bulislaw, 2016-11-07:
The `--profile` option requires a parameter either a profile name (one of debug, default, small) or a path to .json profile file. Adding `--profile` to debug.json file doesn't make any sense, as it's not a compiler/linker switch, it merely includes this json file with compiler options.
Did you try specifying the full path to the debug.json file in your mbed compile line? eg.--profile mbed-os/tools/profiles/debug.json but you need to put a path specific to your file system, also if you're sure the path you're supplying is correct and it still complains about 'permission denied' check access rights to the debug.json file.

EduardPon, 2016-11-07:
Ok, I totally forgot to use the full path. The symbol table is now generated and break points are operational again. 

Thanks for help and quick response...

closed 2016-11-07 by EduardPon