ARMmbed/mbed-os #3215: lp_ticker test is not compatible with unified sleep implementation

stevew817, 2016-11-07:
The lp_ticker test in TESTS/mbed-hal is fundamentally incompatible with the unified sleep API in use by Silicon Labs targets.

At https://github.com/ARMmbed/mbed-os/blob/3a16ca9855528fdb801b11f275ffdca7b5ed9f26/TESTS/mbed_hal/lp_ticker/main.cpp#L97 there is a call to sleep(), which will turn off us_ticker on Silicon Labs platforms, since there is no object initialized that is using the high-power, high-resolution ticker. Therefore, the us_ticker_read function is not suitable for tracking time across sleep calls.

Please change this test to rely on the RTC as an always-on clock, or use us_ticker properly and set it to timeout after the lp_ticker is expected to return.

bridadan, 2016-11-07:
cc @bulislaw

bulislaw, 2016-11-14:
@stevew817 could you provide a patch, that uses the us_ticker properly on Silicon Labs platforms?

closed 2017-10-27 by ChiefBureaucraticOfficer