ARMmbed/mbed-os #3216: Cannot Export with latest mbed-dev

andcor02, 2016-11-07:
Currently I cannot export programs with the latest version of mbed-dev library, 
although older versions (with the older library structure) seem to work. 

1. This is irrespective of platform. 
2. Or toolchain.

The export error is: 'An error occurred during export. Please contact 
support.' 

Regards,

Andrea

0xc0170, 2016-11-07:
cc @screamerbg

andcor02, 2016-11-17:
Any update on this??

@screamerbg @theotherjimmy

bcostm, 2016-11-22:
Same for me... Any news ?

0xc0170, 2016-12-08:
@bcostm @andcor02 Is this still relevant?

andcor02, 2017-01-06:
Can anyone update me on this?

bcostm, 2017-01-09:
Checked today, still the same...

charles1122, 2017-01-09:
yes,failed to export

0xc0170, 2017-01-09:
@theotherjimmy please have a look

theotherjimmy, 2017-01-09:
So, The exporters on the website are very old right now. Really, you are getting lucky if any of them work. The are being updated. In the mean time, you could try importing with mbed-cli and exporting from there.

sg-, 2017-02-04:
@theotherjimmy Are these issues resolved with the latest deployment?

theotherjimmy, 2017-04-25:
I fixed this a while back. Closing.

DaVolfman, 2017-10-18:
You may say that but I'm looking at this exact problem right now or something with identical symptoms.

theotherjimmy, 2017-10-18:
@DaVolfman The uvision exporter does not work on the website right now. Could it just be affecting that? Are you sure it's all targets as well?

DaVolfman, 2017-10-19:
I was trying for sw4stm32 and that didn't work.  Also GCC-ARM (I probably
got the string wrong here) didn't work.  Everything I tried didn't work
with mbed-dev as my library.  I could get an export to complete with
straight mbed, but the project won't build because Eclipse keeps looking
for mbed_config.h in some random temp directory and not finding it.  I
would really like to get this to work in Linux so I don't have to drag
around the desktop replacement laptop running Windows everywhere.

theotherjimmy, 2017-10-19:
@DaVolfman

> I would really like to get this to work in Linux.

Have you tried Mbed CLI? You can use Mbed CLI on Linux to build things. I do it all the time (it's basically the only way I build Mbed OS).

theotherjimmy, 2017-10-19:
> I was trying for sw4stm32 and that didn't work

Dang. I'll make a note to try that one out in the future. It should probably work. That being said, mbed-dev is not the way I would go about doing development. I recommend using Mbed OS 5 (this repo). You will probably find that, with the exception of the RTOS, most of the code can be garbage collected out.

theotherjimmy, 2017-10-23:
@DaVolfman  Were you able to get something working? I'm taking a look into the exporter issue that you reported, and I would l like to get some more information so that I can reproduce the issue.
 * Was this export online? That is from the developer.mbed.org or os.mbed.com websites?
 * Is `mbed-dev` the only library that you are using?
 * What version of `mbed-dev` are you using?
 * What target are you trying to export for?

theotherjimmy, 2017-10-23:
I just took a look about, and it looks like the Online IDE has a "GCC (Code Sourcery)" option, which does nothing, as it's not an exporter. Further, the SW4STM32 does not seem to care about it's [list of supported targets](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/sw4stm32/__init__.py#L36-L257) and allowed me to try to export for LPC824 (which is not in that list). Maybe you ran into both of these issues?

DaVolfman, 2017-10-23:
This export was online.  This is for an Embedded elective and the Mbed
webapp and Uvision 5 are the only tools I'll be getting any external
support on.  In this case mbed-dev was the only library in use in these
projects.  The version of mbed-dev (especially being online) is whatever is
the latest in the repositories.  I'm trying to export for an STM32F446RE
Nucleo board.  The reason mbed-dev was used was to attempt to follow the
instructions on the SW4STM32 website which said to use mbed-src instead of
mbed to avoid any conflict between the library and the setting of the IDE
in the technique for handling floating-point operations (SW4STM32
preferring speed with MBED supposedly preferring code-reuse and
cross-platform compatibility).  Since the mbed-src package appears to be
deprecated in favor of mbed-dev I used that instead.  With either my 446
Nucleo or a 401 Nucleo selected as the target and mbed-dev selected as a
library any export filter/script (terminology?) I have tried to use fails.
These include "Keil Uvision 5", "GCC (ARM Embedded)", "SW4STM32 System
Workbench for STM32", and "Embitz" (because something said the Emblocks
exporter worked and this is the only thing that sounds similar). That's
pretty much the limit to that problem for me.

The instructions from the SW4STM32 website referring to a deprecated
library admittedly does not make me feel warm and fuzzy about the continued
accuracy of those instructions.  Also there is an SW4STM32 export option
now so I've tried exporting these basic GPIO projects using a normal mbed
library and the appropriate filter.  The export completes for this (as well
as Uvision) but I've been unable to get the exported project to build in
SW4STM32 (Uvision's fine) because it's looking for a copy of the
mbed_config.h at some nonexistant path in a trash temp-gibberish
subdirectory name (that incorporates an extra slash in the path which I
doubt is well liked by any filesystem.  This is a different bug altogether.

Honestly if console MBED includes a debugger I will gladly learn to use
it.  Otherwise that's why my entire class is mostly just using the Mbed
library and working in Uvision as soon as possible.

theotherjimmy, 2017-10-23:
Thanks for the detailed explanation! I'm going to look into these failing exports in the next few days. If I'm not mistaken:
 * GCC_ARM - Fails to export for a project using mbed-dev and NUCLEO_F401RE, NUCLEO_F446RE
 * EMbitz, same as GCC_ARM
 * SW4STM32, Garbage temp path for linker script.
 * uVision5, Works

For now, I think that uVision is a good option. I'll let you know when I track down those other bugs.

-------------

More detailed response below:

> follow the instructions on the SW4STM32 website which said to use mbed-src instead of mbed to avoid any conflict between the library and the setting of the IDE in the technique for handling floating-point operations 

Those instructions do indeed sound out of date. I don't recall when the name switch was made. That may give you an indication of how long ago it was, or how short my memory is 😄 

>SW4STM32 preferring speed with MBED supposedly preferring code-reuse and cross-platform compatibility

I think they are referring to `-mfloat-abi=hard` vs `-mfloat-abi=softfp` here. I have yet to see any conclusive evidence that either of these two is faster/more space efficient, so Mbed OS chose `-mfloat-abi=softfp` for compatibility with code compiled with `-mfloat-abi=soft`.

>Since the mbed-src package appears to be deprecated in favor of mbed-dev I used that instead.

Sounds great!

> With either my 446 Nucleo or a 401 Nucleo selected as the target and mbed-dev selected as a library any export filter/script (terminology?) I have tried to use fails.

This is definitely a bug. I'm sorry that you ran into that. A workaround you could try is to use the offline tools, Mbed CLI, to do the export. For example: `mbed export -i sw4stm32 -m nucleo_f446re` should create a project for System Workbench for STM32 from an Mbed CLI project.

Terminology wise, we call them "exporters". They are actually a part of the Mbed OS Tools, and are written in Python.

>The instructions from the SW4STM32 website referring to a deprecated library admittedly does not make me feel warm and fuzzy about the continued accuracy of those instructions. 

Agreed. I can help with project workflow, or call upon someone more knowledgeable if you have questions about something along those lines. Ask away!

> because it's looking for a copy of the mbed_config.h at some nonexistant path in a trash temp-gibberish subdirectory name

That is a bug. I will look into it soon. You could work around it by removing the "trash temp-gibberish", as `mbed_config.h` should be in the root of your project.

>that incorporates an extra slash in the path which I doubt is well liked by any filesystem

Surprisingly, most file systems don't seem to mind. On my linux box for example from mbed-os:
```
# ls ./////
as.uvprojx  cmsis	     doxyfile_options	   doxygen_options.json  events    hal		LICENSE  platform   requirements.txt  targets  tools
BUILD	    CONTRIBUTING.md  DOXYGEN_FRONTPAGE.md  drivers		 features  Jenkinsfile	mbed.h	 README.md  rtos	      TESTS
```
I'm as surprised as your are. I thought this would fail too.

> Honestly if console MBED includes a debugger I will gladly learn to use it.

You can use [pyOCD](https://github.com/mbedmicro/pyOCD) as a gdb-server and GDB. One problem worth mentioning with this setup is that GDB cannot flash binaries created by the Arm Compiler.

>Otherwise that's why my entire class is mostly just using the Mbed library and working in Uvision as soon as possible.

That should work. It may be easiest for you to save the exported zip and host that yourself. Depending on the application, it can be pretty small.

DaVolfman, 2017-11-02:
Some final details: the online exporter seems to be more or less completely
broken.  I get a uvision project that builds, it just has no access to the
pins anymore.  I've got mbed-cli up and running (your website changed and
broke the links to the page where the windows installer could be downloaded
by the way) on multiple systems.  Uvision5 exporting with mbed works
perfectly through mbed-cli.  SW4STM32 doesn't build if exported with the
mbed library (the wrapped functions for memory management aren't found) but
that doesn't matter because it works with the mbed-dev library.  I can now
get a fully-functional IDE with debug support for an STM32 running on
Linux, Windows and almost certainly MacOS and I can probably help other
students get that working.

theotherjimmy, 2017-11-02:
> I get a uvision project that builds, it just has no access to the pins anymore.

Thanks for the bug report. I'll take a look at this in the next few days. What exactly do you mean by "no access to the pins"? Does the IDE's peripheral viewer fail? Are you unable to use pin definitions, such as `PC_7` in C/C++?

> I've got mbed-cli up and running (your website changed and broke the links to the page where the windows installer could be downloaded by the way) on multiple systems. 

@AnotherButler Are you in control of these links? Do you know who is? If so, can we update them to point to the correct destination?

> Uvision5 exporting with mbed works perfectly through mbed-cli.

@DaVolfman That's great to hear! I'm glad something is working for you.

>SW4STM32 doesn't build if exported with the mbed library (the wrapped functions for memory management aren't found) but that doesn't matter because it works with the mbed-dev library.

The fact that it does not work with the mbed library is also a bug. @ajaakko-arm Could you take a look at this? It looks like there might be some hard coded linker flags in the template.

>I can now get a fully-functional IDE with debug support for an STM32 running on Linux, Windows and almost certainly MacOS and I can probably help other students get that working.

If you run into any other snags please let us know. Your feedback so far has been very helpful. Thank you!

AnotherButler, 2017-11-02:
@DaVolfman Thank you for letting us know. When you came across the broken links, were you looking at our documentation for the [Mbed CLI for Windows installer](https://os.mbed.com/docs/v5.6/tools/setup.html#mbed-cli-for-windows-installer)? Or were you somewhere else?

DaVolfman, 2017-11-02:
Honestly I haven't figured it out yet. I won't have time to nail down if
main is even running for a while since my own problems are solved.

DaVolfman, 2017-11-03:
The download link in particular.  I think I was trying to access it right
as it rolled over from 0.41 to 0.42 so the link to the installer went to a
page that didn't exist.

DaVolfman, 2017-11-03:
Either it just got fixed or it's that I was using PWM on pin PB_10 and DAC
on port PA_4 and I forgot to move the wire when changing the method.
Either way export, with Mbed online, to Uvsion 5, with mbed library, is
working right now.  And I learned that PWM on PA_4 isn't a thing while I
was at it.

theotherjimmy, 2017-11-03:
Thanks for the update. It's great to hear that uvision 5 is working for you.

kamenlitchev, 2017-12-10:
Having same issue - RTOS example, trying export for VSCode or GCC (STM32F207)

AnotherButler, 2018-01-09:
@screamerbg @theotherjimmy Could y'all please take a look at this?

0xc0170, 2018-06-15:
Anyone still having this issue ? Please report with details

SenRamakri, 2018-07-23:
Looks like there hasn't been any response to question from @0xc0170 . So Im closing this issue assuming its no longer reproduced. Please re-open or open a new issue if the issue still persists.

closed 2018-07-23 by SenRamakri