ARMmbed/mbed-os #3218: [rtos] thread_stack_main with DEFAULT_STACK_SIZE is too large for some targets

JojoS62, 2016-11-07:
### Description
- Type: Enhancement
- Priority: Major

## Enhancement

**Reason to enhance or problem with existing solution**
The setting of thread_stack_main in RTX_CM_lib.h is very confusing. It is depending on another literaly WORDS_STACK_SIZE, defined in cmsis_os.h. It gets several times multiplied and divided, depending on toolchain and with many exceptions for different targets.

**Suggested enhancement**
A simple default value like 4096 (byte) to fit for most targets with enough RAM:
```
#ifndef (DEFAULT_STACK_SIZE)
#    define DEFAULT_STACK_SIZE 4096
#endif
```

Then this option could be overwritten by passing the define to the compiler, generated from the targets.json or the config system as described in https://github.com/ARMmbed/mbed-os/blob/master/docs/config_system.md
In the config system documentation is the stack size already an example value.

0xc0170, 2016-11-08:
Thanks for tracking this. It was already pointed out partially in https://github.com/ARMmbed/mbed-os/pull/2984.

cc @c1728p9 @bulislaw

JojoS62, 2017-02-08:
a solution is discussed and provided in #3675.

closed 2017-02-08 by JojoS62