ARMmbed/mbed-os #3223: STM32L053-DISCO can't use putc,printf in Keil, EWARM

COMOkr, 2016-11-08:
#include "mbed.h"

DigitalOut green_led(LED1);
DigitalOut red_led(LED2);

Serial a(PA_9, PA_10);

int main() 
{
    green_led = 1;
    
    a.baud(9600);
    red_led = 1;
    
    a.printf("c");
    
    green_led = 0;	
}

On web compiler, It worked normaly
but in exported projects, didn't work

in "a.printf("c");" , program was stopped 
can i have solution?

Platform is STM32L053-DISCO

0xc0170, 2016-11-08:
cc @bcostm @adustm @LMESTM @dbestm

0xc0170, 2016-11-08:
@COMOkr It might help to provide the version numbers (what version of mbed-os you use), plus what expoters - is it uvision or makefile arm gcc , or any other.  As much details as you can provide

bridadan, 2016-11-08:
If you change `a.printf("c");` to `a.printf("c\n");`, does it work? Its possible that the buffers just didn't flush

COMOkr, 2016-11-09:
this solution was't working..

I always update library before making projects
I don't know this version

exporter - uvision 5 , IAR SYSTEMS

bcostm, 2016-11-18:
Issue confirmed.

I have checked this:
- Nucleo_printf example on NUCLEO_L053R8 on mbed IDE (mbed lib v129) --> OK
- Export example to uVision5
- Run example on uVision5 --> Hard Fault !

Analysis on-going...

LMESTM, 2017-05-24:
@COMOkr is this issue still valid ?  Also in case you still have the problem, may you try out wiht RawSerial class instead of Serial one ?

LMESTM, 2017-06-30:
@bcostm this sounds like an issue with exporters rather than a bug in ST HAL ... how should we proceed  ?

bcostm, 2017-06-30:
I really don't know what is the issue. First I need to check again this config `pc.printf + NUCLEO_L053R8 + uVision5 project from mbed IDE` to see if the issue is still present.

theotherjimmy, 2017-06-30:
@bcostm @LMESTM If you think that we should turn of export for targets that don't support mbed OS 5, let me know. I can make that patch, and then we can verify that the export shows "not-supported" and close this issue.

bcostm, 2017-07-03:
> If you think that we should turn of export for targets that don't support mbed OS 5, let me know. 

I don't know if the "mbed export" command has to be used only for mbed OS 5 compliant platforms ? If it is the case, then yes we should turn off this exporter on mbed OS 2 only platforms.

theotherjimmy, 2017-07-10:
@bcostm `mbed export` may be used for mbed OS 2 programs as well as mbed OS 5 programs. The difference here is that exporting a mbed OS 5 project to an mbed OS 2 only board might be causing an issue here. I can disable that, as it's not guaranteed to be supported.

LMESTM, 2017-08-17:
@theotherjimmy @bcostm - the above proposal sounds good to me. This seems indeed a bad idea to allow mbed5 projects on mbed2 only boards.

LMESTM, 2017-08-23:
>  I can disable that, as it's not guaranteed supported.

 @bcostm  Ok with that ?
@theotherjimmy Could you go ahead with this proposal?

bcostm, 2017-08-23:
OK

theotherjimmy, 2017-08-23:
Yup. PR in a moment.

theotherjimmy, 2017-08-23:
@LMESTM @bcostm @0xc0170 Please see #4961

PR #4961 (2017-09-27): Check for mbed 5 support on export
Resolves #3223
closed 2017-09-27 by theotherjimmy