ARMmbed/mbed-os #3224: Error while trying to commit/publish online IDE

andcor02, 2016-11-08:
Hi,

So this seems to be a reoccurring issue in that some programs/libraies for some users seem not to able to commit/publish changes. I have had this happen to myself, only on a certain number of programs, they still cannot be committed. "There has been a problem while committing your project".

In short the error:

- Cannot commit certain programs/libraries (This is permanent)
- Not sure if this mbed OS specific

Regards,

Andrea, mbed

0xc0170, 2016-11-08:
cc @theotherjimmy @screamerbg

sarahmarshy, 2016-11-11:
I think this is a known issue when you try to commit a repo that has a git repo inside it. @theotherjimmy  @screamerbg You can reproduce this by adding the atmel-rf driver from https://github.com/ARMmbed/atmel-rf-driver to any project, and then trying to commit/publish the project.

closed 2017-02-04 by sg-