ARMmbed/mbed-os #3226: "mbed compile" fails with "list index out of range"

infinnovation, 2016-11-08:
### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**Toolchain version:**
4_9-2015q3

**mbed-cli version:** 
0.9.10

**meed-os sha:**
e2617cc

**Expected behavior**
Failed compilation should not propagate to next compilation attempt

**Actual behavior**
In some circumstances, a failed compilation results in the next "mbed compile" reporting
```
...
Scan: mbed
Scan: env
[ERROR] list index out of range
[mbed] ERROR: "python" returned error code 1.
```

**Steps to reproduce**
Ah, you've got me there!  I haven't been able to determine the exact circumstances.  What I have found is that somehow a .d file may be left empty after a failed compilation.  In one example, the error was a missing include file referenced by main.cpp, but the file BUILD/K64F/GCC_ARM/mbed-os/events/equeue/equeue_mbed.d was left empty.

When an empty .d file is present, GCC.parse_dependencies in tools/toolschains/gcc,py falls over when trying to access the first line of the file.

0xc0170, 2016-11-08:
cc @theotherjimmy

theotherjimmy, 2016-11-08:
So your steps to reproduce would be to take a clean repo and create an empty `.d` file.

theotherjimmy, 2016-11-08:
Reproduced artificially by:
```bash
$ echo -n > BUILD/K64F/gcc_arm/main.d
```

theotherjimmy, 2016-11-09:
@infinnovation could you review #3229 ?

PR #3229 (2016-11-15): [Tools - Toolchains] Allow dependency parsing to fail, gracefully continuing
## Description
If, say, you compile with gcc_arm and kill it at just the right time you will have an empty dependency file. This empty dependency file will cause a python traceback in dependency parsing, preventing you from compiling until you next clean.

This PR lets that weird corner case compile.

## Status
**READY**

## Reviews
- [x] @screamerbg (toolchains modified)
- [ ] @infinnovation (issue reporter)

should resolve #3226 

## Reproducing the issue

We can simulate the very precise timing needed to reproduce  [... cut]
closed 2016-11-15 by sg-