ARMmbed/mbed-os #3228: NanoInterface.cpp asserts when non LWM2M server can be found.

EduardPon, 2016-11-08:
In case no LWM2M can be found, after some time (>5 minutes), the function NanostackSocket::data_available triggers an assert because the variable mode==SOCKET_MODE_OPENED. At this point rxBufChain != NULL. 

```
bool NanostackSocket::data_available()
{
    nanostack_assert_locked();
    MBED_ASSERT((SOCKET_MODE_DATAGRAM == mode) ||
                (SOCKET_MODE_CONNECTING == mode) ||
                (SOCKET_MODE_STREAM == mode));
    return (NULL == rxBufChain) ? false : true;
}
```

```
Thread #1 1 (Thread mode) [core: 0] (Suspended : Breakpoint)	
	NanostackSocket::data_available() at NanostackInterface.cpp:337 0x93438	
	NanostackInterface::socket_recvfrom() at NanostackInterface.cpp:858 0x93d92	
	UDPSocket::recvfrom() at UDPSocket.cpp:113 0x95636	
	pal_plat_receiveFrom() at pal_plat_network.cpp:396 0x96876	
	pal_receiveFrom() at pal_network.c:235 0x95fdc	
	M2MConnectionHandlerPimpl::receive_handler() at m2mconnectionhandlerpimpl.cpp:565 0x5560c	
	connection_event_handler() at m2mconnectionhandlerpimpl.cpp:52 0x54bd0	
	eventOS_scheduler_dispatch_event() at event.c:292 0x6e784	
	eventOS_scheduler_run() at event.c:317 0x6e7a2	
	event_loop_thread() at ns_event_loop.c:76 0x6cae8	
	() at rt_CMSIS.c:906 0x98d50	
```

0xc0170, 2016-11-08:
cc @SeppoTakalo @kjbracey-arm

kjbracey-arm, 2016-11-08:
So it's complaining because you're calling recvfrom on a socket which apparently hasn't had anything else done to it?

It wants it to be bound or something? Looks like it thinks that maybe the state should be "DATAGRAM"?

Which it should have set if anyone had ever called bind or sendto?

So it seems like a vaguely plausible sort of application sanity check on the face of it. As it's a bit weird to call recvfrom in that state - your socket should never have actually received anything.

Although, apparently you have received something. Interesting.

EduardPon, 2016-11-09:
So the LWM2M client want to send a registration message to aabb::30. But aabb::30 is not on the network present. All initiated receives data fail (NSAPI_ERROR_WOULD_BLOCK   = -3001). Comment of enumeration:  /*!< no data is not available but call is non-blocking */

So who initiated this ESocketReadytoRead event, see call stack:
```
Thread #1 1 (Thread mode) [core: 0] (Suspended : Breakpoint)	
	M2MConnectionHandlerPimpl::send_receive_event() at m2mconnectionhandlerpimpl.cpp:82 0x54c2a	
	socket_event_handler() at m2mconnectionhandlerpimpl.cpp:94 0x54c5a	
	mbed::Callback<void ()>::function_call<void (*)()> at Callback.h:726 0x46a60	
	mbed::Callback<void ()>::call() const at Callback.h:656 0x541d4	
	mbed::Callback<void ()>::operator()() const at Callback.h:662 0x956c0	
	UDPSocket::event() at UDPSocket.cpp:152 0x95ce0	
	Socket::close() at Socket.cpp:66 0x94c3a   <<<<<<<<<<< CLOSE INITIATES AN EVENT SEND????	
	pal_plat_close() at pal_plat_network.cpp:448 0x96f2a	
	pal_close() at pal_network.c:259 0x96606	
	M2MConnectionHandlerPimpl::close_socket() at m2mconnectionhandlerpimpl.cpp:702 0x55ad6	
	M2MConnectionHandlerPimpl::dns_handler() at m2mconnectionhandlerpimpl.cpp:245 0x5504c	
	connection_event_handler() at m2mconnectionhandlerpimpl.cpp:64 0x54c00	
	eventOS_scheduler_dispatch_event() at event.c:292 0x6ea08	
	eventOS_scheduler_run() at event.c:317 0x6ea26	
	event_loop_thread() at ns_event_loop.c:76 0x6cd6c	
	() at rt_CMSIS.c:906 0x99304	
```
Some logging:
```
[DBG ][mClt]: send_data()
[DBG ][mClt]: send_handler()
[DBG ][nsif]: socket_sendto(socket=0x2000bed0) sock_id=2, ret=918
[DBG ][nsif]: socket_callback() sock=2, event=144, interface=0, data len=918
[DBG ][nsif]: SOCKET_TX_DONE, 918 bytes sent
[DBG ][mClt]: receive_handler()
[DBG ][nsif]: socket_recvfrom(socket=0x2000bed0) sock_id=2, ret=-3001
[DBG ][perObjMan]: Current heap: 63211  Max heap size: 66067
[DBG ][ip6r]: Sending unicast NS for: aabb::30
[DBG ][icmp]: Build NA
[DBG ][perObjMan]: Current heap: 63211  Max heap size: 66067
[DBG ][ip6r]: Sending unicast NS for: fe80::58c:47ab:7411:7561
[DBG ][perObjMan]: Current heap: 63211  Max heap size: 66067
[DBG ][perObjMan]: Current heap: 63211  Max heap size: 66067
[DBG ][mClt]: stop_listening()
[DBG ][perObjMan]: Current heap: 62219  Max heap size: 66067
[DBG ][perObjMan]: Current heap: 62219  Max heap size: 66067
[DBG ][mClt]: stop_listening()
[DBG ][mClt]: resolve_server_address()
[DBG ][mClt]: IP Address aa:bb:00:00:00:00:00:00:00:00:00:00:00:00:00:30
[DBG ][mClt]: close_socket() - IN
[DBG ][nsif]: socket_attach(socket=0x2000bed0) sock_id=2
[DBG ][nsif]: socket_close(socket=0x2000bed0) sock_id=2
**SOCKET_MODE_CLOSED**
[DBG ][mClt]: close_socket() - OUT
[DBG ][mClt]: init_socket - IN
[DBG ][mClt]: Interface count: 1
[DBG ][m6LEth]: getOwnIpAddress()
[DBG ][mClt]: Interface name:
[DBG ][mClt]: Interface no: 0
[DBG ][mClt]: init_socket - port 51910      <<<< IS THIS AN INVALID PORT NR? 
[DBG ][sck ]: Socket id 3 allocated
**SOCKET_MODE_OPENED**
[DBG ][nsif]: socket_open() socket=0x2000bed0, sock_id=3, ret=0
[DBG ][nsif]: socket_attach(socket=0x2000bed0) sock_id=3
[DBG ][mClt]: init_socket - OUT
[DBG ][mClt]: start_listening_for_data()
[DBG ][mClt]: send_data()
[DBG ][mClt]: receive_handler() 
>>>>> NanostackSocket::data_available   [ERR ][nsif]: ASSERT IS TRIGGERED !   <<<<<<
```

kjbracey-arm, 2016-11-09:
You said rxBufChain != NULL - it actually is NULL?

My general feeling here is that the assert should just be removed. Don't see what purpose it's serving.

If we wanted any state enforcement checks, they should be actual error returns in recvfrom and recv.

Presumably though send_data() near the end of the trace didn't actually send any data? socket_sendto would have changed the state to DATAGRAM, silencing the assert.

Don't see the point of going into OPENED/DATAGRAM distinction myself. A datagram socket should just go straight into DATAGRAM. There's no point starting in OPENED and changing to DATAGRAM later.

Whereas the OPENED->CONNECTING->STREAM->CLOSED state progression could be potentially useful.

EduardPon, 2016-11-10:
> You said rxBufChain != NULL 
That's correct.

Could it be that the **event generate** in the Socket::close() at Socket.cpp @ line 66 is little bit strange? Primary this results in a M2MConnectionHandlerPimpl::send_receive_event().

kjbracey-arm, 2016-11-10:
The callback to users is deliberately vague - it just indicates "something" has happened. So as such we shouldn't be terribly surprised if people respond to it by attempting a read. An assert is unreasonable.

Tthe event() appears to be desirable to signal the "wake from blocking" semaphore - less certain about whether it should be generating a callback. Seems semi-reasonable, I guess. The socket being closed is "something happening".

On the other hand, anyone actually relying on getting an event when that happens and then attempting something with the now-recycled socket ID seems like they would be opening themselves up to race condition hell.

So maybe the user callback should be cleared before the event().

@geky ?

kjbracey-arm, 2016-11-10:
Still not quite clear why you have pending data in rxBufChain in this case, but it's probably not relevant.

EduardPon, 2016-11-10:
The rxBufChain != NULL but points maybe to outdated data and nothing is pending.

kjbracey-arm, 2016-11-10:
That's contradictory, implying either a bug or a misunderstanding. The rxBufChain is the received data that has not yet been read by the application. Nanostack refuses to store data, so the abstraction layer is required to copy and buffer the data synchronously inside the RX data callbacks it gets from Nanostack.

If it points to anything, it should be something the application hasn't read yet. See NanostackSocket::data_copy_and_free.

Maybe it received 2 packets, and decided to close after reading and not liking the first one?

EduardPon, 2016-11-14:
@geky, any comments?

sg-, 2017-02-04:
@EduardPon Update on the status of this?

EduardPon, 2017-02-06:
No change. I'm still using a modified version that has replaced the assert with a tr_error.

kjbracey-arm, 2017-02-06:
This bug will almost certainly vanish in 5.4 - the Nanostack interface code is radically different in this area, and it loses the rxbufChain stuff altogether. (It relies on newly-added socket buffering inside Nanostack instead).

Suggest rechecking then.

closed 2017-10-27 by ChiefBureaucraticOfficer