ARMmbed/mbed-os #3234: mbed_lwip_bringup() with static IP

YixiaoLi, 2016-11-09:
I tried to initialize an EthernetInterface with static IP but it didn't work.
```c
auto eth = new EthernetInterface();
eth->set_network("10.0.10.1", "255.255.255.0", "10.0.10.1");
eth->connect();
```
After doing some research, I did find a workaround by modifying `mbed_lwip_bringup()`.
Inside `mbed_lwip_bringup()`, `lwip_connected ` will only be set when no IP address is specified.
I modified the code as follows and now it works.
```c
    // If doesn't have address
    if (!mbed_lwip_get_ip_addr(true, &lwip_netif)) {
        ret = sys_arch_sem_wait(&lwip_netif_has_addr, 15000);
        if (ret == SYS_ARCH_TIMEOUT) {
            return NSAPI_ERROR_DHCP_FAILURE;
        }
//        lwip_connected = true; // original code
    }
    lwip_connected = true; // workaround
```
I wonder if it is a bug or something, and what is the correct approach to assign a static IP address?

jupe, 2016-11-09:
Looks that you are trying to use same interface and gateway addresses. Probably that wouldn't work anyway.

hakanernam, 2016-11-09:
I have the same issue with the same kind of workaround. Problem is not the particular addresses used. Somewhere in the same file:

```
static nsapi_error_t mbed_lwip_socket_open(nsapi_stack_t *stack, nsapi_socket_t *handle, nsapi_protocol_t proto)
{
    // check if network is connected
    if (!lwip_connected) {
        return NSAPI_ERROR_NO_CONNECTION;
}
...
```

if static IP is used, lwip_connected is not set to true, therefore you get NSAPI_ERROR_NO_CONNECTION trying to open a socket connection.  I think lwip_connected in the case of static IP should be set based on the result of `EthernetInterface::set_network` call

kdeme, 2016-11-15:
I came across the same issue and I am using the same fix.

I think that having the `lwip_connected = true` line on the original location is a mistake. 
The if clause is checking if it already has an address and if not it will wait (until timeout). However if it already has an address, it can continue immediately, but the lwip_connected boolean should be set (moving `lwip_connected = true` out of the if clause). 

This usually will be the case with static IP setting, if it has been properly set before doing the `eth->connect()` call. In theory it could happen also with DHCP, but I guess in practice it never will as `dhcp_start()` is ran right before this if clause.

I think you could also argue about the name of the boolean. Currently it is only checking if there is an address set (whether static of via DHCP), hence the name lwip_connected is not the best in my opinion.

0xc0170, 2016-11-16:
cc @geky

okoshi3, 2016-11-22:
Wifi also has the same problem. (I am using ODIN - W2.)
In my recognition, since Wifi and Ether have different headers as topics,
I think that another modification is necessary.
I made the following corrections for example(mbed-os-example-wifi).

```
    int ret = wifi.set_network("172.20.10.10", "255.255.255.240", "172.20.10.1");
    if (ret != 0) {printf("\r\nset_network error\r\n");return -1;}

    ret = wifi.set_credentials(MBED_CONF_APP_WIFI_SSID, MBED_CONF_APP_WIFI_PASSWORD, NSAPI_SECURITY_WPA_WPA2);
    if (ret != 0) {printf("\r\nset_credentials error\r\n");return -1;}

    ret = wifi.connect();
    if (ret != 0) {printf("\r\nconnect error\r\n");return -1;}

    printf("Success\r\n\r\n");
    printf("MAC: %s\r\n", wifi.get_mac_address());
    printf("IP: %s\r\n", wifi.get_ip_address());
    printf("Netmask: %s\r\n", wifi.get_netmask());
    printf("Gateway: %s\r\n", wifi.get_gateway());
    printf("RSSI: %d\r\n\r\n", wifi.get_rssi());

    http_demo(&wifi);

    wifi.disconnect();
```

The result is below.
The connection is successful, but we have not received http request.
Connect with DHCP It works as expected.

```
Scan:
 ....
 Network: iPhone xxxxx secured: WPA2 BSSID: xx:xx:xx:xx:xx:xx RSSI: -17 Ch: 11
 ....

 15 networks available.
 
 Connect
 Success
 
 MAC: d4:ca:6e:70:47:68
 IP: 172.20.10.10
 Netmask: 255.255.255.240
 Gateway: 172.20.10.1
 RSSI: -17
 
 Sending HTTP request to www.arm.com...
 sent -3004 [GET / HTTP/1.1]
```
 
Here's how it works with DHCP.

```
Connect
Success

MAC: d4:ca:6e:70:47:68
IP: 172.20.10.9
Netmask: 255.255.255.240
Gateway: 172.20.10.1
RSSI: -26

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]
```
Where is the problem?
I can not get a hand when I see it in library objects ...

geky, 2017-01-04:
Thanks for raising the issue. @YixiaoLi's is right that the `lwip_connected = true` statement needs to be moved.

It looks like the initialization of lwip for static ip addresses doesn't set a dns server. It does for ipv6, but it looks like the path for ipv4 relies on dns servers from dhcp. I think the issue is the same on the Odin board. Working on a patch to fix this.

With @YixiaoLi you should be able to connect to literal ip addresses (not domain names).

geky, 2017-01-04:
Hopefully #3526 resolves all these problems, let us know if there are still issues.

closed 2017-01-09 by sg-