ARMmbed/mbed-os #3235: [Platform] limited type deduction of callback generator

pan-, 2016-11-09:
### Description
- Type: Enhancement 

----------------------------------------------------------------
## Enhancement

The `callback` function and all its overloads is a generator of `Callback` objects. Its main purpose is to avoid typing of the type of the `Callback` the user wants and avoid too many callbacks objects on the stack: 

```c++
void foo(const Callback<void(int, int)>&);

class Bar { 
public:
  void handler(int, int);
};

static Bar bar;

void qux() { 
  // inject the callback in foo with callback generator.
  // the temporary callback will be destroyed right after the function invocation
  foo(mbed::callback(&bar, &Bar::handler));

  // inject the callback in foo by constructing it manually
  // the callback object will be destroyed at the end of the scope
  { 
    Callback<void(int, int)> bar_handler(&bar, &Bar::handler);
    foo(bar_handler);
  }
}
```

To generate the proper `Callback` type, the `callback` function relies on template type deduction but the type deduction can fail in sometimes for valid use case:

```c++
void foo(const Callback<void(int, int)>&);

class Bar { 
public:
  void handler(int, int);
};

class BarDerived { 
public:
  void qux() { 
     // 1. error no match for callback(BarDerived*, void(Bar::*)(int, int))
     foo(mbed::callback(this, &BarDerived::handler));

     // 2. error void Bar::handler(int, int) protected in this context.
     foo(mbed::callback(this, &Bar::handler));

     // 3. error void Bar::handler(int, int) protected in this context.
     foo(mbed::callback(static_cast<Bar*>(this), &Bar::handler));

     // 4. correct
     foo(mbed::callback(static_cast<Bar*>(this), &BarDerived::handler)); 
  }
};

```

**Suggested enhancement**

* Points `2` and `3` can't be addressed by code but can't be addressed by documentation. Since it is very difficult to track the actual error when the compiler spit hundreds of errors on the command line, it would be interesting to have a list of the common errors users might encounter when they use callback.

* Point `1` can be addressed by relaxing the constraints of the types in input of the callback function, For the example presented earlier, the type of the expected callback has this form:
```c++
template<typename T, typename R, typename A0>
Callback<R(A0)> callback(T *obj, R (T::*func)(A0)) {
    return Callback<R(A0)>(obj, func);
}
```
Since C++ allows derived objects to call parents member functions, it is not safe to assume that the same `T` is applicable for the object and the pointer to the member function. 

One solution would be to add another type in the template parameter list to untie the object type from the pointer to member function type: 

```c++
template<typename T, typename U, typename R, typename A0>
Callback<R(A0)> callback(T *obj, R (U::*func)(A0)) {
    return Callback<R(A0)>(obj, func);
}
```

Of course the change should be reflected in the `Callback` class to make it work. Any type error should be catch at a latter point (where `(obj->*func)(...)` is instantiated). This change doesn't endanger the type safety of the callback class.

pan-, 2016-11-09:
@SeppoTakalo

This is the issue you reported yesterday, please leave any needed comments.

@geky @0xc0170 

The size of the callback related facilities is another concern. The flat organization was supposed to makes it easier to understand but when you got a three thousands line long errors due to the overloads explosions, it is really not a pleasant experience as a user. 
Any plan to refine the callback facilities ?

geky, 2016-11-09:
Thanks for looking into this! Your solution looks like the best solution, so I went ahead and put together a pr to bring it in: https://github.com/ARMmbed/mbed-os/pull/3244

It does sounds like the callback class needs some form of documentation page.

> The size of the callback related facilities is another concern. The flat organization was supposed to makes it easier to understand but when you got a three thousands line long errors due to the overloads explosions, it is really not a pleasant experience as a user.
Any plan to refine the callback facilities ? 

I agree, it's not the best experience. Any suggestions for refining the class? How could we improve the user experience without losing the functionality? It would also be nice to keep the implementation simple for contributors, but this may be a secondary concern.

pan-, 2016-11-10:
> It does sounds like the callback class needs some form of documentation page.

Yes, maybe we can open another issue listing what's missing in Callback documentation.

> Any suggestions for refining the class? How could we improve the user experience without losing the functionality? It would also be nice to keep the implementation simple for contributors, but this may be a secondary concern.

I think the following items can be improved: 
* most of the internal details can be factored out in private/detail implementation files (and base class too!), it can also reduce the template bloat. 
* A function like attach makes less sense today because the class implement correct copy assignment. Basically `f.attach(new_function)` is equivalent to `f = callback(new_function)`.
* Most of the overloads can be removed involving function pointer (not pointer to member function!) can be removed. 

Unfortunately since the order of arguments is not consistent between pointer to member function (`this` then the pointer to member function) and  function pointer (function then first argument) , it will be hard to factorize to a single entry point the callback function. 

It can still be done but it will be hard to understand for non seasoned c++ programmers.

0xc0170, 2016-11-16:
@pan- Has this been fixed via #3244? Any outstanding work to be done?

pan-, 2016-11-16:
No, the documentation issue and the simplification of the code remains valid items. Maybe it can be tracked in another issue.

sg-, 2017-02-04:
@pan- If docs are needed please open an new issue

closed 2017-02-04 by sg-