ARMmbed/mbed-os #3236: wifiInterface.h callback? and error in method documentation for scan

lauri-piikivi, 2016-11-09:
- Type: Bug 
- Priority: ?

---------------------------------------------------------------
## Bug 1

The API has the include callback.h, but is it used ?

## Bug 2

the documentation for scan method seems incomplete and from old version, timeout is not a parameter and ap is not parameter name. 
    /** Scan for available networks
     *
     *  The scan will 
     *  If the network interface is set to non-blocking mode, scan will attempt to scan
     *  for WiFi networks asynchronously and return NSAPI_ERROR_WOULD_BLOCK. If a callback
     *  is attached, the callback will be called when the operation has completed.
     *
     * @param  ap       Pointer to allocated array to store discovered AP
     * @param  count    Size of allocated @a res array, or 0 to only count available AP
     * @param  timeout  Timeout in milliseconds; 0 for no timeout (Default: 0)
     * @return          Number of entries in @a, or if @a count was 0 number of available networks, 
     *                  negative on error
     *                  see @a nsapi_error
     */
    virtual nsapi_size_or_error_t scan(WiFiAccessPoint *res, nsapi_size_t count) = 0;

0xc0170, 2016-11-09:
Thanks for reporting, we will fix it

cc @bulislaw

bulislaw, 2016-11-10:
As for one, it's not right now we sort of developed async version calls, but then removed it for further discussion, we can remove it for now.

And for two that looks like merging issue.

Will fix them today.

PR #3247 (2016-11-10): WiFi: Minor fixes to docs and includes

## Description
Fix WiFi issues pointed in #3236 

## Status
**READY**

@0xc0170 @lauri-piikivi 

Closes #3236
closed 2016-11-10 by adbridge