ARMmbed/mbed-os #3246: [lwip] Zero or wrong error returned from failed TCPSocket.connect()

infinnovation, 2016-11-09:
### Description
- Type: Bug
- Priority: ?

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**mbed-os sha:**
a1c0840

**Expected behavior**
1) TCPSocket.connect() returns error is connection is refused
2) TCPSocker.connect() returns appropriate error code for e.g. unreachable host

**Actual behaviour**
1) TCPSocket.connect() returns 0
2) TCPSocket.connect() returns NSAPI_ERROR_DEVICE_ERROR, implying e.g. a hardware problem

**Steps to reproduce**
See [example program](https://developer.mbed.org/users/infinnovation/code/connect-test/)

0xc0170, 2016-11-10:
cc @geky @c1728p9

0xc0170, 2016-11-16:
> cc @geky @c1728p9

Please have a look at this issue

c1728p9, 2016-11-22:
It looks like these errors aren't getting properly propagated up from lwip. These errors will need to be converted to nsapi errors here:
https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_LWIP/lwip-interface/lwip_stack.c#L536

closed 2017-01-10 by adbridge