ARMmbed/mbed-os #3249: Mbed Lib Revision 128: "mbed compile" fail with default_build not found

aracosta1, 2016-11-10:
### Description
- Type: Bug 

**Target**
NRF51_DK, RBLAB_BLENANO, maybe all

**Toolchain:**
GCC_ARM

**mbed-cli version:** 
(`mbed --version`)
0.9.10

**meed-os sha:**
https://github.com/ARMmbed/mbed-os/commit/e2617cc0e17f5c3fc2bae6a589c9bcfd3d1a717b
tag: mbed_lib_rev128

**Steps to reproduce**

I downloaded the mbed_blinky examples and imported mbed library at revision 127:25aea2a3f4e3
Set target NRF51_DK, toolchain GCC_ARM
"mbed compile" build run correctly

Updated the mbed library to revision 128:9bcdf88f62b0
"mbed compile" terminate with the following error :

    [ERROR] Attribute 'default_build' not found in target 'NRF51_DK'
    [mbed] ERROR: "python" returned error code 1.
    [mbed] ERROR: Command "python -u E:\Mbed\mbed_blinky\.temp\tools\make.py -t GCC_ARM -m NRF51_DK --source . --build .\BUILD\NRF51_DK\GCC_ARM" in "E:\Mbed\mbed_blinky"

I see inside targets.json "default_build" has been replaced by "default_lib" and this break the compile script. There are some steps to be done before recompileng the application with the new lib version ?

0xc0170, 2016-11-10:
@theotherjimmy Can you please have a look?

0xc0170, 2016-11-16:
> @theotherjimmy Can you please have a look?

Bump?

theotherjimmy, 2016-11-16:
Sorry, I looked into this and forgot to respond. The format for targets.json has changed in mbed 128. The tools used by classic projects are a different version than is used inside the codebase here. The classic tools do not accept both `default_build` and `default_lib`, and only accept `default_build`.

theotherjimmy, 2016-11-16:
https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools/

aracosta1, 2016-11-18:
Will the tools be updated ?

theotherjimmy, 2016-11-18:
I'm working on it.

0xc0170, 2016-12-07:
> I'm working on it.

Is this issue still relevant? Can be closed?

aracosta1, 2016-12-12:
It's still open, the tools fail to compile at the moment.

adbridge, 2016-12-13:
@aracosta1 Have you tried the latest version (130) to see if it is still a problem there ?

aracosta1, 2016-12-13:
The problem is still present in version 130.

theotherjimmy, 2017-01-12:
@0xc0170 This can be closed when https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools/pull-request/1 is merged.

0xc0170, 2017-01-19:
> @0xc0170 This can be closed when https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools/pull-request/1 is merged.

Any  luck merging that PR ? I can see its still open.

theotherjimmy, 2017-01-19:
I will be getting push access to that repo soon. I'll let you know when that PR's changes get it.

sg-, 2017-02-04:
@aracosta1 @theotherjimmy has this been resolved?

theotherjimmy, 2017-02-27:
Yes. Closing.

closed 2017-02-27 by theotherjimmy