ARMmbed/mbed-os #3255: [TARGET_NORDIC] restore the S110 softdevice on TARGET_NRF5

aracosta1, 2016-11-10:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

**With the TARGET_MCU_NRF51822_UNIFIED only S130 is available**

**Restore the possibility to select the S110 softdevice**

**Pros**
The S110 uses less memory and in most application the Central role is never used, so using the S110 the application will have about 2KB more ram to use.

**Cons**
Maybe too much work for a small gain ?

pan-, 2016-11-11:
This won't be possible I'm afraid. The target `TARGET_MCU_NRF51822_UNIFIED` is based on Nordic SDK v11 and use the softdevice v2.0. 

With the version 2.0 of the softdevice, the S110 has disappeared, instead the softdevice allows the user to configure the number of connection and the BLE roles used during the initialization. This change allows the user to adjust the softdevice memory usage to its needs and it become possible to use less memory if the device doesn't act as a central, exactly like the S110. 

Unfortunately, memory provisioning still occur at link time and changing the softdevice configuration means changing the linker script to provision the right amount of memory for the softdevice. 

@nvlsianpu have created script generating the linker script based on user configuration, maybe he can provides more information. As far as I know, the remaining issue with was the integration of the linker script generation in the build system.

aracosta1, 2016-11-12:
Thank you for the explanation.

closed 2016-11-12 by aracosta1

nvlsianpu, 2016-11-14:
You can estimate softdevice memory consumption using this script :https://github.com/nvlsianpu/mbed/blob/memo_provisionibg/hal/targets/cmsis/TARGET_NORDIC/TARGET_NRF5/script/gen_ld.py
See https://github.com/nvlsianpu/mbed/blob/memo_provisionibg/hal/targets/cmsis/TARGET_NORDIC/TARGET_NRF5/script/redmefirst.md.txt for HOWTO USE IT.

Unfortunately you must modify memory map if you want to apply changes (e.g. https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF51822_UNIFIED/device/TOOLCHAIN_GCC_ARM/TARGET_MCU_NORDIC_32K/NRF51822.ld#L6).