ARMmbed/mbed-os #3259: Inconsistent header file naming. critical.h should be mbed_critical.h

adbridge, 2016-11-11:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

critical.c was renamed to mbed_critical.c during the file renaming phase, however it's header , critical.h was not renamed. This now looks inconsistent.

**Suggested enhancement**
Rename /platform/critical.h to /platform/mbed_critical.h

**Pros**
Maintain a consistent codebase.

**Cons**
none

bridadan, 2017-01-12:
@adbridge after having a discussion with @geky, it's mostly because deprecating header files is harder than deprecating .c files. However we both agree that the rename would definitely help with consistency.

The way forward we suggest is move the existing contents of `critical.h` to `mbed_critical.h` (there other header files that we could update at the same time). Then create a `critical.h` that includes the `mbed_critical.h` but also add an `MBED_DEPRECATED_SINCE` (https://github.com/ARMmbed/mbed-os/blob/master/platform/toolchain.h#L245) macro to that file to warn users to switch to `mbed_critical.h`.

This would also unfortunately affect mbed 2 builds since this is a shared file in mbed OS. So we'd have to take that into consideration.

@sg- do you have any thoughts on this?

sg-, 2017-01-13:
mbed.h is the supported interface for API usage so I wouldn't be much afraid of this. We can fix all the errors in the mbed OS tree. The deprecated macro and temporary existence of the older filename sounds like a good path forward.

0xc0170, 2017-01-16:
> mbed.h is the supported interface for API usage so I wouldn't be much afraid of this. We can fix all the errors in the mbed OS tree. The deprecated macro and temporary existence of the older filename sounds like a good path forward.

+1, that's what we have done previously for instance in mbed-drivers module.

closed 2017-02-23 by sg-