ARMmbed/mbed-os #3260: MBED OS for STM32F412ZG MCU

andcor02, 2016-11-14:
Just a quick query, when is Mbed OS available for this [platform](https://www.element14.com/community/docs/DOC-83063/l/stm32-nucleo-144-development-board-with-stm32f412zg-mcu-supports-arduino-st-zio-and-morpho-connectivity)?

On this site it mentions being mbed enabled, but it currently isn't?

When is the ETA for this? @screamerbg

0xc0170, 2016-11-16:
cc @adustm @bcostm

bcostm, 2016-11-17:
The SDK development is not planned for the moment by ST. But anyone can start it and send a pull request :) Take example with other F4 devices. The F412 specific files (startup files, register mapping file, ...) are present in the ST CubeF4 fw package. And many SDK files are common for all F4 devices, so the development should be "easy".

closed 2016-12-06 by andcor02

bcostm, 2016-12-06:
PR #3366