ARMmbed/mbed-os #3261: ASYNC API usage/documentation

Sissors, 2016-11-14:
### Description
- Type: Enhancement / Question
- Related issue: `#abc`
- Priority: Minor

---------------------------------------------------------------

As example the from the I2C API, although the same is relevant for others. From the I2C.h file:

```
/** Start non-blocking I2C transfer.
     *
     * @param address   8/10 bit I2c slave address
     * @param tx_buffer The TX buffer with data to be transfered
     * @param tx_length The length of TX buffer in bytes
     * @param rx_buffer The RX buffer which is used for received data
     * @param rx_length The length of RX buffer in bytes
     * @param event     The logical OR of events to modify
     * @param callback  The event callback function
     * @param repeated Repeated start, true - do not send stop at end
     * @return Zero if the transfer has started, or -1 if I2C peripheral is busy
     */
    int transfer(int address, const char *tx_buffer, int tx_length, char *rx_buffer, int rx_length, const event_callback_t& callback, int event = I2C_EVENT_TRANSFER_COMPLETE, bool repeated = false);
```

From looking through the source files, if I understand it correctly, callback is called when one of the events from the event argument happened (such as a transfer complete). So the question, is this correct?

If this is indeed correct, then why does the Doxygen say: @param event     The logical OR of events to modify. How should I read this? What does it mean? Wouldn't it make sense to say something like: "The events which trigger the callback". (Or the logical OR of the events which trigger the callback).

Next one, why are these events hidden in the HAL files? I assume because the target specific driver files need to also use them, so they can't be in the C++ .h files? While in general I am against having duplicate data/information/code, in this case wouldn't it be better to include the ones you can use in the Doxygen? Or if possible make a Doxygen link to the file which does include it.

bridadan, 2016-11-14:
cc @0xc0170

LMESTM, 2016-11-14:
my 2 cents
* @param event     The logical OR of events to modify
was maybe meant to be 
* @param event     The logical OR of events to notify 
?

closed 2017-10-27 by ChiefBureaucraticOfficer