ARMmbed/mbed-os #3264: NUCLEO_F429ZI Test Failures

bridadan, 2016-11-14:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
NUCLEO_F429ZI

**Toolchain:**
GCC_ARM, ARM, and IAR

**Toolchain version:**
4.9.3 2015q2

**mbed-cli version:** 
`0.9.10`

**meed-os sha:**
`cb930e748230d9ccb287629d2eb603baaf63e88a`

**STLink version:**
V2.J28.M16

**Expected behavior**
All tests pass!

**Actual behavior**
Not all tests pass!

The following tests fail:

### tests-mbed_hal-lp_ticker
```
[1479155484.16][CONN][RXD] >>> Running case #4: '5s lp_ticker'...
[1479155484.20][CONN][INF] found KV pair in stream: {{__testcase_start;5s lp_ticker}}, queued...
[1479155489.10][CONN][RXD] :58::FAIL: Values Not Within Delta 100000 Expected 5000000 Was 4833776
[1479155489.15][CONN][INF] found KV pair in stream: {{__testcase_finish;5s lp_ticker;0;1}}, queued...
[1479155489.22][CONN][RXD] >>> '5s lp_ticker': 0 passed, 1 failed with reason 'Test Cases Failed'
```

### tests-mbed_drivers-lp_timeout
```
1479155504.99][CONN][RXD] >>> Running case #4: '5sec LowPowerTimeout'...
1479155505.04][CONN][INF] found KV pair in stream: {{__testcase_start;5sec LowPowerTimeout}}, queued...
1479155509.93][CONN][RXD] :96::FAIL: Values Not Within Delta 100000 Expected 5000000 Was 4821860
1479155509.98][CONN][INF] found KV pair in stream: {{__testcase_finish;5sec LowPowerTimeout;0;1}}, queued...
1479155510.07][CONN][RXD] >>> '5sec LowPowerTimeout': 0 passed, 1 failed with reason 'Test Cases Failed'
```

cc @bcostm @adustm

jeromecoutant, 2016-11-17:
Hi
For information, this test is OK on my side :-)
But I notice that RTC timer (used for low power timers) are currently  configured with LSI clock, whereas NUCLEO board have LSE.
Check is on going
Regards,

jeromecoutant, 2016-11-23:
You should integrate https://github.com/ARMmbed/mbed-os/pull/3316 and https://github.com/ARMmbed/mbed-os/pull/3317
and retry
Thx

bridadan, 2016-12-13:
Looks like this issue is now resolved on master, thanks @jeromecoutant!

bridadan, 2016-12-13:
Reopening this. I'm still seeing an issue with the `tests-mbed_drivers-lp_timeout` test, though now it's a mutex failing to initialize. And it's only occurring when I use the CI machines to build the test, which use a specific version of GCC_ARM. It works ok on my local machine, but I'm using a different version of GCC_ARM than the CI machines. I ran across this exact type of failure in #3319, maybe these two issues are related?

It may just be that we need to use a different version of GCC_ARM, though I'd prefer to understand the problem first. It seems odd that it's only occurring for STM32F4 targets when the test uses the low power timer!

bcostm, 2017-03-03:
Rechecked with mbed-os-5.4.0. lp_timeout test is OK.
```
+-----------------------+---------------+----------------------------------+-------------------------------------+--------+--------+--------+--------------------+
| target                | platform_name | test suite                       | test case                           | passed | failed | result | elapsed_time (sec) |
+-----------------------+---------------+----------------------------------+-------------------------------------+--------+--------+--------+--------------------+
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-c_strings     | C strings: %e %E float formatting   | 1      | 0      | OK     | 0.08               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-c_strings     | C strings: %f %f float formatting   | 1      | 0      | OK     | 0.06               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-c_strings     | C strings: %g %g float formatting   | 1      | 0      | OK     | 0.06               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-c_strings     | C strings: %i %d integer formatting | 1      | 0      | OK     | 0.06               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-c_strings     | C strings: %u %d integer formatting | 1      | 0      | OK     | 0.06               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-c_strings     | C strings: %x %E integer formatting | 1      | 0      | OK     | 0.08               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-c_strings     | C strings: strpbrk                  | 1      | 0      | OK     | 0.04               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-c_strings     | C strings: strtok                   | 1      | 0      | OK     | 0.04               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-dev_null      | tests-mbed_drivers-dev_null         | 1      | 0      | OK     | 15.57              |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-echo          | Echo server: x16                    | 1      | 0      | OK     | 2.2                |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-echo          | Echo server: x32                    | 1      | 0      | OK     | 4.42               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-echo          | Echo server: x64                    | 1      | 0      | OK     | 8.68               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-flashiap      | FlashIAP - init                     | 1      | 0      | OK     | 0.04               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-flashiap      | FlashIAP - program                  | 1      | 0      | OK     | 2.58               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-flashiap      | FlashIAP - program errors           | 1      | 0      | OK     | 0.06               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-generic_tests | Basic                               | 1      | 0      | OK     | 0.04               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-generic_tests | Blinky                              | 1      | 0      | OK     | 0.03               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-generic_tests | C++ heap                            | 1      | 0      | OK     | 0.1                |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-generic_tests | C++ stack                           | 1      | 0      | OK     | 0.15               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-lp_timeout    | 1ms LowPowerTimeout                 | 1      | 0      | OK     | 0.05               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-lp_timeout    | 1sec LowPowerTimeout                | 1      | 0      | OK     | 1.04               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-lp_timeout    | 1sec LowPowerTimeout from deepsleep | 1      | 0      | OK     | 1.07               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-lp_timeout    | 1sec LowPowerTimeout from sleep     | 1      | 0      | OK     | 1.07               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-lp_timeout    | 500us LowPowerTimeout               | 1      | 0      | OK     | 0.05               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-lp_timeout    | 5sec LowPowerTimeout                | 1      | 0      | OK     | 5.06               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-race_test     | class init race                     | 1      | 0      | OK     | 0.65               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-race_test     | function init race                  | 1      | 0      | OK     | 0.62               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-rtc           | RTC strftime                        | 1      | 0      | OK     | 10.52              |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-stl_features  | STL std::equal                      | 1      | 0      | OK     | 0.05               |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-stl_features  | STL std::sort abs                   | 1      | 0      | OK     | 0.05               |
[... cut ...]
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-timeout       | Timers: toggle on/off               | 1      | 0      | OK     | 11.06              |
| NUCLEO_F429ZI-GCC_ARM | NUCLEO_F429ZI | tests-mbed_drivers-wait_us       | Timers: wait_us                     | 1      | 0      | OK     | 11.15              |
+-----------------------+---------------+----------------------------------+-------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 37 OK
```

Can we close this issue ?

bridadan, 2017-03-06:
I also could not reproduce this issue, so I'll close it!

closed 2017-03-06 by bridadan