ARMmbed/mbed-os #3273: RZ_A1H Tests Failing

bridadan, 2016-11-16:
### Description
- Type: Bug
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
RZ_A1H

**Toolchain:**
GCC_ARM, ARM, and IAR

**Toolchain version:**
GCC_ARM - 4.9.3 20141119
ARM - ARM Compiler 5.06 update 1
IAR - V7.80.1.11864/W32 for ARM

**mbed-cli version:** 
`0.9.10`

**meed-os sha:**
`9d8ec61df5e7442bbde12beabe744ed3a3e13486`

**DAPLink version:**
0221 according to DETAILS.TXT, but I think this is running CMSIS-DAP

**Expected behavior**
All tests pass! 

**Actual behavior**
Not all tests pass!

**Steps to reproduce**
`mbed test -m RZ_A1H -t GCC_ARM -v -n tests-mbed_drivers-race_test,tests-mbedmicro-rtos-mbed-malloc,tests-mbedmicro-rtos-mbed-threads`

**Failing tests**

**NOTE:** These failures were consistent across all toolchains

### tests-mbed_drivers-race_test
```
[1479316325.57][HTST][INF] host test detected: default_auto
[1479316325.58][CONN][INF] found KV pair in stream: {{__testcase_count;2}}, queued...
[1479316325.61][CONN][RXD] >>> Running 2 test cases...
[1479316325.66][CONN][INF] found KV pair in stream: {{__testcase_name;function init race}}, queued...
[1479316325.70][CONN][RXD]
[1479316325.70][CONN][INF] found KV pair in stream: {{__testcase_name;class init race}}, queued...
[1479316325.75][CONN][RXD] >>> Running case #1: 'function init race'...
[1479316325.79][CONN][INF] found KV pair in stream: {{__testcase_start;function init race}}, queued...
[1479316325.80][CONN][RXD] Running function race test
[1479316325.82][CONN][RXD] TestClass ctor start
[1479316326.36][CONN][RXD] TestClass ctor end
[1479316345.63][HTST][INF] test suite run finished after 20.10 sec...
[1479316345.64][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1479316345.65][HTST][INF] CONN exited with code: 0
[1479316345.65][HTST][INF] No events in queue
[1479316345.65][HTST][INF] stopped consuming events
[1479316345.65][HTST][INF] host test result(): None
[1479316345.65][HTST][WRN] missing __exit event from DUT
[1479316345.65][HTST][WRN] missing __exit_event_queue event from host test
[1479316345.65][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1479316345.65][HTST][INF] calling blocking teardown()
[1479316345.65][HTST][INF] teardown() finished
[1479316345.65][HTST][INF] {{result;timeout}}
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test on hardware with target id: 550002210FB3423630F7860B
mbedgt: test suite 'tests-mbed_drivers-race_test' .................................................... TIMEOUT in 31.87 sec
        test case: 'function init race' .............................................................. ERROR in 0.00 sec
[... 80 lines cut ...]
[1479316644.29][CONN][INF] found KV pair in stream: {{__testcase_name;Testing thread self terminate}}, queued...
[1479316644.30][CONN][RXD]
[1479316644.35][CONN][RXD] >>> Running case #1: 'Testing single thread'...
[1479316644.39][CONN][INF] found KV pair in stream: {{__testcase_start;Testing single thread}}, queued...
[1479316683.96][HTST][INF] test suite run finished after 40.61 sec...
[1479316683.97][CONN][INF] received special even '__host_test_finished' value='True', finishing
[1479316683.98][HTST][INF] CONN exited with code: 0
[1479316683.98][HTST][INF] No events in queue
[1479316683.98][HTST][INF] stopped consuming events
[1479316683.98][HTST][INF] host test result(): None
[1479316683.98][HTST][WRN] missing __exit event from DUT
[1479316683.98][HTST][WRN] missing __exit_event_queue event from host test
[1479316683.98][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1479316683.98][HTST][INF] calling blocking teardown()
[1479316683.98][HTST][INF] teardown() finished
[1479316683.98][HTST][INF] {{result;timeout}}
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'TIMEOUT'
mbedgt: test on hardware with target id: 550002210FB3423630F7860B
mbedgt: test suite 'tests-mbedmicro-rtos-mbed-threads' ............................................... TIMEOUT in 52.73 sec
        test case: 'Testing single thread' ........................................................... ERROR in 0.00 sec
mbedgt: all tests finished!
```

cc @toyowata, @TomoYamanaka

TomoYamanaka, 2016-11-21:
@bridadan 

Thank you for giving me the comment. And sorry for this late reply.

I don't understand well the occurs.
Does this errors cause by the version of the automatic test tool (mbed-cli, mbed-greentea, mbed-ls etc)?
Because we passed these three tests for mbed-os 5 in Aug.
After that, I made a minor fix, but I don't think I changed the content that affects these tests.

cc @toyowata

bridadan, 2016-11-21:
@TomoYamanaka Thanks for your reply. I was just doing a general check up on this platform and saw there were a few test failures and I thought I'd notify you.

The `tests-mbed_drivers-race_test` test was added to the code base on Aug 30 in this PR: https://github.com/ARMmbed/mbed-os/pull/2562. This was added by @c1728p9, maybe he has some insight as to what's the problem?

The `tests-mbedmicro-rtos-mbed-malloc` test was also added by @c1728p9 so I might let him comment on that.

The `tests-mbedmicro-rtos-mbed-threads` may be related to the two tests above, so let's focus on those two at the moment.

adbridge, 2016-11-21:
@c1728p9 Russ can you comment please?

c1728p9, 2016-11-21:
I took a look at the test tests-mbedmicro-rtos-mbed-malloc in depth and found a couple of problems that are causing these failures:

1. Cortex A is using an older version of RTX which is causing problems. Specifically, rt_tsk_delete isn't releasing mutexes held by a task that exits or is terminated. This causes the function thread.join() to get stuck. Updating to RTX 4.8 will fix this problem. Worst case, just rt_tsk_delete could be patched.

2. With ARMCC there is a stack overflow in this test. The define THREAD_STACK_SIZE is too small for cortex A which is causing this overflow.

3. Malloc locking is missing for GCC_ARM. The functions __rtos_malloc_lock, __rtos_malloc_unlock and __rtos_env_lock need to be implemented in mbed-os\rtos\rtx\TARGET_CORTEX_A\RTX_CM_lib.h for GCC.

TomoYamanaka, 2016-11-22:
@bridadan 
Thanks for your reply. I understood that those was a newly added tests.

@c1728p9 
Thanks you for your commnets. 
My understandind on your feedback is that these test error occur because "Cortex-A" is using an older version of "RTX" and some functions are nothing.
Is it right ?

c1728p9, 2016-11-22:
Hi @TomoYamanaka, the old version of RTX is 1 of the 3 problems. To pass testing all 3 need to be fixed. The old version of RTX is causing problems because rt_tsk_delete() has a bug in it - it does not release mutexes held by the terminated task.

TomoYamanaka, 2016-11-24:
@c1728p9
 Thanks you for your commnets. I was misunderstanding.
Renesas will consider measures for three problems.

TomoYamanaka, 2016-12-01:
@c1728p9 
> 2.With ARMCC there is a stack overflow in this test. The define THREAD_STACK_SIZE is too small for 
> cortex A which is causing this overflow.

I recognize this THREAD_STACK_SIZE macro defined to the following file. It is right ?
&nbsp;&nbsp;TESTS\mbedmicro-rtos-mbed\malloc\main.cpp

This file is a test file for <code> tests-mbedmicro-rtos-mbed-malloc </code>. 
Therefore, I think that it is necessary to "#ifdef" the definition of THREAD_STACK_SIZE for each target.
For example, as in main.cpp below.
&nbsp;&nbsp;TESTS\mbedmicro-rtos-mbed\basic\main.cpp

Is it possible to modify it by an ARM member who is a test case creator?

c1728p9, 2016-12-02:
Hi @TomoYamanaka I created this test so feel free to modify it. I would recommend an "#if defined(__CORTEX_A9)" and inside it set the stack to a larger value. I think cortex-A in general will require more stack space. If you would like I can make this change.

TomoYamanaka, 2016-12-02:
@c1728p9 
Thank you for your prompt reply and suggestions.

I will be very helpful if you change it. Please make this change.
As is the other main.cpp for test, I think that this stack size will be DEFAULT_STACK_SIZE(= 2048).

c1728p9, 2016-12-03:
Hi @TomoYamanaka, I created #3362 to increase the stack size.

TomoYamanaka, 2016-12-05:
@c1728p9
 Thank you for your support.

TomoYamanaka, 2016-12-06:
Now Renesas is resolving this issue.
After fix it, I will push PR.
If anything, please let me ask a question again.

Best regards

TomoYamanaka, 2016-12-20:
I pushed PR for this issue.
Reference PR is #3470 .

bridadan, 2016-12-21:
Hi @TomoYamanaka, thanks for all the work! The tests look good when testing with your PR, however in CI we also enable heap and stack stats by running the following commands:

```
mbed test -m RZ_A1H -t GCC_ARM -DMBED_HEAP_STATS_ENABLED=1 --clean --compile
mbed test -m RZ_A1H -t GCC_ARM --run -v
```

I'm receiving an mbed assertion from the memory metrics library for each test. I'm leaving for the holidays so I can't debug this at the moment, but perhaps you could look into it with @c1728p9 ?

TomoYamanaka, 2016-12-21:
Hi @c1728p9 
cc @bridadan 

Thank you for your comments.

I knew this test for the first time. So I will try this test as soon.
Can you tell me detail about what kind of problem is happening?

TomoYamanaka, 2016-12-22:
@c1728p9 

I ran the above two commands. But I did not receive an mbed assertion, did not find like the message in test log.
Also, There was another mbed board (NUCLEO) at hand, so we compared PEACH's test log and it's test log. However, I could not find the noticeable difference.

If you can reproduce in your environment, I would like to let me share test log. 
Could you upload it here?

bridadan, 2016-12-27:
@TomoYamanaka Thanks for trying this locally. Could you tell me the version of GCC_ARM you're using? I'd like to reproduce your results locally as well. Here's my environment:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

TomoYamanaka, 2016-12-28:
Hi @bridadan 

Thank you for your reply.
Here is my GCC_ARM version:
```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
<br>
It seems the same environment as yours.

bridadan, 2016-12-29:
Thanks @TomoYamanaka, unfortunately I'm out of office for the holidays and don't have access to a RZ_A1H, so I can't test at this time. I will do so when I return! I will try in other environments as well, it may just be something wrong with my machine.

bridadan, 2017-01-06:
Hi @TomoYamanaka, it turns out it wasn't the HEAP_STATS but it was the STACK_STATS that was causing the issue.

Could you please run the following commands your machine and see if you get a similar error?

In a fresh workspace:
```
git clone https://github.com/ARMmbed/mbed-os
cd mbed-os
git checkout e7361ebc442de7a7252a4e191043471e9763550b
mbed test -m RZ_A1H -t GCC_ARM -DMBED_STACK_STATS_ENABLED=1 --clean --compile
mbed test -m RZ_A1H -t GCC_ARM --run -n tests-mbed_drivers-stl_features -v
```

This was the output from the test command:
```
C:\Users\bridan01\Documents\dev\m_mbed\mbed-os>mbed test -m RZ_A1H -t GCC_ARM --run -n tests-mbed_drivers-stl_features -v
[mbed] Working path "C:\Users\bridan01\Documents\dev\m_mbed\mbed-os" (program)
[mbed] Exec "mbedgt --test-spec C:\Users\bridan01\Documents\dev\m_mbed\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json -n tests-mbed_drivers-stl_features -V" in C:\Users\bridan01\Documents\dev\m_mbe
d\mbed-os
mbedgt: greentea test automation tool ver. 1.2.4
mbedgt: test specification file 'C:\Users\bridan01\Documents\dev\m_mbed\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Users\bridan01\Documents\dev\m_mbed\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | RZ_A1H        | RZ_A1H[0]            | COM318      | D:          | 550002211D9E1CC022DAD8FD |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'RZ_A1H' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | RZ_A1H        | RZ_A1H[0]            | COM318:9600 | D:          | 550002211D9E1CC022DAD8FD |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: test case filter (specified with -n option)
        test filtered in 'tests-mbed_drivers-stl_features'
mbedgt: running 1 test for platform 'RZ_A1H' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m RZ_A1H -p COM318:9600 -f "BUILD/tests/RZ_A1H/GCC_ARM/TESTS/mbed_drivers/stl_features/stl_features.bin" -d D: -C 2 -c shell -t 550002211D9E1CC022DAD8FD -e "TESTS
\host_tests"
mbedgt: mbed-host-test-runner: started
[1483719737.44][HTST][INF] host test executor ver. 1.1.6
[1483719737.44][HTST][INF] copy image onto target...
[1483719737.44][COPY][INF] Waiting up to 60 sec for '550002211D9E1CC022DAD8FD' mount point (current is 'D:')...
        1 file(s) copied.
[1483719746.63][HTST][INF] starting host test process...
[1483719746.97][CONN][INF] starting connection process...
[1483719746.97][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1483719746.97][CONN][INF] initializing serial port listener...
[1483719746.97][PLGN][INF] Waiting up to 60 sec for '550002211D9E1CC022DAD8FD' serial port (current is 'COM318')...
[1483719746.99][HTST][INF] setting timeout to: 60 sec
[1483719747.74][SERI][INF] serial(port=COM318, baudrate=9600, timeout=0.01)
[1483719747.75][SERI][INF] reset device using 'default' plugin...
[1483719748.00][SERI][INF] waiting 1.00 sec after reset
[1483719749.00][SERI][INF] wait for it...
[1483719749.00][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1483719749.00][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1483719749.01][CONN][INF] sending preamble '1d63eb76-d299-4b82-83c1-e3dae1a94186'
[1483719749.01][SERI][TXD] {{__sync;1d63eb76-d299-4b82-83c1-e3dae1a94186}}
[1483719749.14][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1483719749.19][CONN][INF] found SYNC in stream: {{__sync;1d63eb76-d299-4b82-83c1-e3dae1a94186}} it is #0 sent, queued...
[1483719749.20][HTST][INF] sync KV found, uuid=1d63eb76-d299-4b82-83c1-e3dae1a94186, timestamp=1483719749.188000
[1483719749.21][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1483719749.21][HTST][INF] DUT greentea-client version: 1.3.0
[1483719749.23][CONN][INF] found KV pair in stream: {{__timeout;5}}, queued...
[1483719749.23][HTST][INF] setting timeout to: 5 sec
[... 63 lines cut ...]
+----------------+---------------+---------------------------------+---------+--------------------+-------------+
| RZ_A1H-GCC_ARM | RZ_A1H        | tests-mbed_drivers-stl_features | TIMEOUT | 17.61              | shell       |
+----------------+---------------+---------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 1 TIMEOUT
mbedgt: test case report:
[... cut ...]
mbedgt: test case results: 4 OK
mbedgt: completed in 18.93 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error code 1.
```

This is the error line causing the TIMEOUT:
```
[1483719750.46][CONN][RXD] mbed assertation failed: ret, file: C:\Users\bridan01\Documents\dev\m_mbed\mbed-os\features\frameworks\greentea-client\source\greentea_metrics.cpp, line 150
```

Can you confirm you see something similar? If you don't see it with GCC_ARM, try using the ARM or IAR toolchains. Thanks for your patience!

TomoYamanaka, 2017-01-10:
@bridadan 

Thank you for your commnets.

As a result of executing in your method, I could reproduce the similar phenomenon.
<br>

```
C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os>mbed test -m RZ_A1H -t GCC_ARM --run -n tests-mbed_drivers-stl_features -v
[mbed] Working path "C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os" (program)
[mbed] Exec "mbedgt --test-spec C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json -n tests-mbed_drivers-stl_features -V" in C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os
mbedgt: greentea test automation tool ver. 1.2.4
mbedgt: test specification file 'C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | RZ_A1H        | RZ_A1H[0]            | COM4        | D:          | 5500022104BE4F4D3BFA8B70 |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'RZ_A1H' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | RZ_A1H        | RZ_A1H[0]            | COM4:9600   | D:          | 5500022104BE4F4D3BFA8B70 |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: test case filter (specified with -n option)
        test filtered in 'tests-mbed_drivers-stl_features'
mbedgt: running 1 test for platform 'RZ_A1H' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m RZ_A1H -p COM4:9600 -f "BUILD/tests/RZ_A1H/GCC_ARM/TESTS/mbed_drivers/stl_features/stl_features.bin" -d D: -C 2 -c shell -t 5500022104BE4F4D3BFA8B70 -e "TESTS\host_tests"
mbedgt: mbed-host-test-runner: started
[1484040593.42][HTST][INF] host test executor ver. 1.1.5
[1484040593.42][HTST][INF] copy image onto target...
[1484040593.42][COPY][INF] Waiting up to 60 sec for '5500022104BE4F4D3BFA8B70' mount point (current is 'D:')...
        1 個のファイルをコピーしました。
[1484040602.59][HTST][INF] starting host test process...
[1484040603.15][CONN][INF] starting connection process...
[1484040603.16][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1484040603.16][CONN][INF] initializing serial port listener...
[1484040603.16][SERI][INF] serial(port=COM4, baudrate=9600, timeout=0.01)
[1484040603.16][PLGN][INF] Waiting up to 60 sec for '5500022104BE4F4D3BFA8B70' serial port (current is 'COM4')...
[1484040603.17][HTST][INF] setting timeout to: 60 sec
[1484040603.35][SERI][INF] reset device using 'default' plugin...
[1484040603.60][SERI][INF] waiting 1.00 sec after reset
[1484040604.60][SERI][INF] wait for it...
[1484040604.61][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1484040604.61][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1484040604.61][CONN][INF] sending preamble 'bb682d52-b483-46d2-9046-89001f92a5ae'
[1484040604.61][SERI][TXD] {{__sync;bb682d52-b483-46d2-9046-89001f92a5ae}}
[1484040604.74][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1484040604.79][CONN][INF] found SYNC in stream: {{__sync;bb682d52-b483-46d2-9046-89001f92a5ae}} it is #0 sent, queued...
[1484040604.79][HTST][INF] sync KV found, uuid=bb682d52-b483-46d2-9046-89001f92a5ae, timestamp=1484040604.792000
[1484040604.81][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1484040604.82][HTST][INF] DUT greentea-client version: 1.3.0
[1484040604.83][CONN][INF] found KV pair in stream: {{__timeout;5}}, queued...
[1484040604.83][HTST][INF] setting timeout to: 5 sec
[1484040604.87][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1484040604.88][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1484040604.88][HTST][INF] host test setup() call...
[1484040604.88][HTST][INF] CALLBACKs updated
[1484040604.88][HTST][INF] host test detected: default_auto
[1484040604.89][CONN][INF] found KV pair in stream: {{__testcase_count;4}}, queued...
[1484040604.92][CONN][RXD] >>> Running 4 test cases...
[1484040604.95][CONN][INF] found KV pair in stream: {{__testcase_name;STL std::equal}}, queued...
[1484040604.99][CONN][INF] found KV pair in stream: {{__testcase_name;STL std::transform}}, queued...
[... 49 lines cut ...]
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.0787849009
mbedgt: test suite report:
[... cut ...]
| RZ_A1H-GCC_ARM | RZ_A1H        | tests-mbed_drivers-stl_features | STL std::transform    | 1      | 0      | OK     | 0.04               |
+----------------+---------------+---------------------------------+-----------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 4 OK
mbedgt: completed in 18.87 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error code 1.
```
<br>
Does this TIMEOUT message mean that there is not enough stack size to set in the test ?

bridadan, 2017-01-10:
@TomoYamanaka The TIMEOUT is actually because the device halted with an `mbed_assertion` message. The test passed until it got to reporting the memory statistics, as you can see here in the log:

```
[1484040606.08][CONN][RXD] mbed assertation failed: ret, file: C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os\features\frameworks\greentea-client\source\greentea_metrics.cpp, line 150
```

This points to this line: https://github.com/ARMmbed/mbed-os/blob/master/features/frameworks/greentea-client/source/greentea_metrics.cpp#L150

@c1728p9 Do you have suggestions on where to begin debugging this assert?

c1728p9, 2017-01-10:
This assert indicates that the stack info queue was empty when it wasn't expected to be.  This could occur [here](https://github.com/ARMmbed/mbed-os/blob/9e24117b686d70614227ffe961a955ba4cf91097/features/frameworks/greentea-client/source/greentea_metrics.cpp#L97) if ```enqeue_thread_info``` returned without actually enqueuing anything. Maybe check [here](https://github.com/ARMmbed/mbed-os/blob/9e24117b686d70614227ffe961a955ba4cf91097/features/frameworks/greentea-client/source/greentea_metrics.cpp#L125) and below to see if this function returned early.

TomoYamanaka, 2017-01-13:
Hi @c1728p9、@bridadan

Thank you for your comments, and sorry for this late reply.
I will debug based on your advice.

TomoYamanaka, 2017-01-13:
Hi @c1728p9、@bridadan

I found the cause of this assert.
Thread information is acquired by calling `_osThreadGetInfo`() in `enqeue_thread_info` function,
The result of `osThreadInfoStackMax` becomes `osErrorResource`, and an error occurs in [here](https://github.com/ARMmbed/mbed-os/blob/9e24117b686d70614227ffe961a955ba4cf91097/features/frameworks/greentea-client/source/greentea_metrics.cpp#L140).
Cortex-A RTX does not have stack init so the maximum stack usage can not be acquired.
Therefore, it seems that `osErrorResource` is set. reference is [here](https://github.com/ARMmbed/mbed-os/commit/c31929669265b51abd2d3a5102a161fdcf77108d#diff-fc8b921e2bb4ee84adac6242857fd3c0R891).

As `enqeue_thread_info` function of `greentea_metrics.cpp`, in the case of Cortex-A, is it possible to regard status as OK even if it is `osErrorResource` ?

c1728p9, 2017-01-13:
Hi @TomoYamanaka, it looks like the changes [here](https://github.com/ARMmbed/mbed-os/pull/3470/files#diff-f9b52d04089f04f4bbc1765f2576353dR85) added stack init, so if you want you should be able implement osThreadInfoStackMax (just copy the cortex-m version).  If that is a problem or you don't want to implment it, you could wrap [this](https://github.com/ARMmbed/mbed-os/blob/9e24117b686d70614227ffe961a955ba4cf91097/features/frameworks/greentea-client/source/greentea_metrics.cpp#L97) in a check to "if (queue->empty())" to fix the problem.

TomoYamanaka, 2017-01-20:
@c1728p9 

Thanks you for your comments, and sorry for this late reply.

I am trying with the method that you suggested now. 
I will contact you as I know something.

TomoYamanaka, 2017-01-23:
Hi @bridadan @c1728p9 

> it looks like the changes here added stack init, so if you want you should be able implement osThreadInfoStackMax (just copy the cortex-m version). 

As a result of trying with the method that you suggested, I could get **SUCCESS** in the test case of `tests-mbed_drivers-stl_features`,  not **TIMEOUT**.

The Code that I changed  is here.
- rt_CMSIS.c&nbsp;&nbsp;&nbsp;&nbsp;`svcThreadGetInfo () `function
&nbsp;&nbsp;&nbsp;&nbsp;Implement `if (osThreadInfoStackMax == info) {` (just copy the [cortex-M](https://github.com/ARMmbed/mbed-os/commit/52fd7b6864bf70952eba7c7aa5083af9ae77fdab#diff-833d2ec0a0af4b1627b3f1519701b23cR824)). 
- RTX_Conf_CA.c&nbsp;&nbsp;&nbsp;&nbsp; `#ifndef OS_STKINIT`
&nbsp;&nbsp;&nbsp;&nbsp;Implement `OS_STKINIT` macro (just copy the [Cortex-M](https://github.com/ARMmbed/mbed-os/commit/52fd7b6864bf70952eba7c7aa5083af9ae77fdab#diff-2ab06f62c3ba840e95e98024c2014688R139))

In addition, test result after code changed is here.
```
C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os>mbed test -m RZ_A1H -t GCC_ARM --run -n tests-mbed_drivers-stl_features -v
[mbed] Working path "C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os" (program)
[mbed] Exec "mbedgt --test-spec C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json -n tests-mbed_drivers-stl_features -V" in C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os
mbedgt: greentea test automation tool ver. 1.2.4
mbedgt: test specification file 'C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Work\Mission\mbed\GR-PEACH\SRC\New3test_pass\mbed-os\BUILD\tests\RZ_A1H\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | RZ_A1H        | RZ_A1H[0]            | COM3        | D:          | 550002211B6B4CF3242F88CE |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'RZ_A1H' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | RZ_A1H        | RZ_A1H[0]            | COM3:9600   | D:          | 550002211B6B4CF3242F88CE |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: test case filter (specified with -n option)
        test filtered in 'tests-mbed_drivers-stl_features'
mbedgt: running 1 test for platform 'RZ_A1H' and toolchain 'GCC_ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m RZ_A1H -p COM3:9600 -f "BUILD/tests/RZ_A1H/GCC_ARM/TESTS/mbed_drivers/stl_features/stl_features.bin" -d D: -C 2 -c shell -t 550002211B6B4CF3242F88CE -e "TESTS\host_tests"
mbedgt: mbed-host-test-runner: started
[1485155658.08][HTST][INF] host test executor ver. 1.1.5
[1485155658.08][HTST][INF] copy image onto target...
[1485155658.08][COPY][INF] Waiting up to 60 sec for '550002211B6B4CF3242F88CE' mount point (current is 'D:')...
        1 個のファイルをコピーしました。
[1485155667.24][HTST][INF] starting host test process...
[1485155667.52][CONN][INF] starting connection process...
[1485155667.52][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1485155667.52][CONN][INF] initializing serial port listener...
[1485155667.52][SERI][INF] serial(port=COM3, baudrate=9600, timeout=0.01)
[1485155667.52][PLGN][INF] Waiting up to 60 sec for '550002211B6B4CF3242F88CE' serial port (current is 'COM3')...
[1485155667.53][HTST][INF] setting timeout to: 60 sec
[1485155667.67][SERI][INF] reset device using 'default' plugin...
[1485155667.92][SERI][INF] waiting 1.00 sec after reset
[1485155668.93][SERI][INF] wait for it...
[1485155668.93][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1485155668.93][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1485155668.93][CONN][INF] sending preamble 'da6f1c45-b5f5-4a20-b732-0580b1fbe12e'
[1485155668.93][SERI][TXD] {{__sync;da6f1c45-b5f5-4a20-b732-0580b1fbe12e}}
[1485155669.06][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1485155669.11][CONN][INF] found SYNC in stream: {{__sync;da6f1c45-b5f5-4a20-b732-0580b1fbe12e}} it is #0 sent, queued...
[1485155669.11][HTST][INF] sync KV found, uuid=da6f1c45-b5f5-4a20-b732-0580b1fbe12e, timestamp=1485155669.107000
[1485155669.14][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1485155669.14][HTST][INF] DUT greentea-client version: 1.3.0
[1485155669.15][CONN][INF] found KV pair in stream: {{__timeout;5}}, queued...
[1485155669.15][HTST][INF] setting timeout to: 5 sec
[... cut ...]
| RZ_A1H-GCC_ARM | RZ_A1H        | tests-mbed_drivers-stl_features | STL std::sort greater | 1      | 0      | OK     | 0.05               |
| RZ_A1H-GCC_ARM | RZ_A1H        | tests-mbed_drivers-stl_features | STL std::transform    | 1      | 0      | OK     | 0.05               |
+----------------+---------------+---------------------------------+-----------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 4 OK
mbedgt: completed in 14.31 sec
```

If there is no problem in this changes, I will PR this.
I'm looking forward to contacting from you.

bridadan, 2017-01-23:
@TomoYamanak I think those changes should be fine, thanks for looking at this! Please submit them as a PR and we will review further.

TomoYamanaka, 2017-01-23:
@bridadan 

Thank you for your quick reply.
I will PR this change soon.

bridadan, 2017-02-03:
Fixed by #3694, thanks for all of your work @TomoYamanaka!

closed 2017-02-03 by bridadan