ARMmbed/mbed-os #3275: Unable to compile Blinky for LPC1347

grbd, 2016-11-17:
### Description
- Type: Bug
- Priority: Minor

Hi I noticed that trying to compile the BlinkyTest against the LPC1347 target fails with a `.bss' will not fit in region `RAM' error
I recognize that its not a supported platform but I suspect something might not be configured correctly in relation to the heap, but I'm unsure how to fix it.
It looks like the mbed-os library itself compiles okay against this target but linking in the code fails

---------------------------------------------------------------
## Bug

**Target LPC1347**

**Toolchain: GCC_ARM**

**Toolchain version: 4.9 2015q3**

**mbed-cli version: 0.9.10** 

**mbed-os sha: 79abaab**

**Expected behavior**
compile without linkage failure

**Actual behavior**
```
Building project BlinkyTest1 (LPC1347, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
Link: BlinkyTest1
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: ./BUILD/LPC1347/GCC_ARM/BlinkyTest1.elf section `.bss' will not fit in region `RAM'
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region RAM overflowed with stack
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 1492 bytes
collect2.exe: error: ld returned 1 exit status
[ERROR] c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: ./BUILD/LPC1347/GCC_ARM/BlinkyTest1.elf section `.bss' will not fit in region `RAM'
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region RAM overflowed with stack
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 1492 bytes
collect2.exe: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\make.py -t GCC_ARM -m LPC1347 --source . --build .\BUILD\LPC1347\GCC_ARM" in "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1"
```

verbose
```
D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1>mbed compile -v
[mbed] Working path "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1" (library)
[mbed] Exec "python -u D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\make.py -t GCC_ARM -m LPC1347 --source . --build .\BUILD\LPC1347\GCC_ARM -v" in D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1
Building project BlinkyTest1 (LPC1347, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
[DEBUG] Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTOOLCHAIN_object -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LPC13XX -DTARGET_LPC1347 -DTARGET_RELEASE -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1479345132.82
Link: BlinkyTest1
[DEBUG] Link: C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin\arm-none-eabi-gcc @.\BUILD\LPC1347\GCC_ARM\.link_files.txt
[DEBUG] Return: 1
[DEBUG] Errors: c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: ./BUILD/LPC1347/GCC_ARM/BlinkyTest1.elf section `.bss' will not fit in region `RAM'
[DEBUG] Errors: c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region RAM overflowed with stack
[DEBUG] Errors: c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 1492 bytes
[DEBUG] Errors: collect2.exe: error: ld returned 1 exit status
Traceback (most recent call last):
  File "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\make.py", line 293, in <module>
    toolchain))
  File "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\build_api.py", line 456, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\toolchains\__init__.py", line 993, in link_program
    self.link(elf, r.objects, r.libraries, r.lib_dirs, r.linker_script)
  File "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\hooks.py", line 34, in wrapper
    res = function(t_self, *args, **kwargs)
  File "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\toolchains\gcc.py", line 237, in link
    self.default_cmd(cmd)
  File "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\toolchains\__init__.py", line 1021, in default_cmd
    raise ToolException(_stderr)
ToolException: c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: ./BUILD/LPC1347/GCC_ARM/BlinkyTest1.elf section `.bss' will not fit in region `RAM'
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region RAM overflowed with stack
c:/program files (x86)/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 1492 bytes
collect2.exe: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1\mbed-os\tools\make.py -t GCC_ARM -m LPC1347 --source . --build .\BUILD\LPC1347\GCC_ARM -v" in "D:\SourceControl\GitRepos.Appst\mbed.visualmicro.platform\Examples\BlinkyTest1"
```

**Steps to reproduce**

mbed import https://github.com/ARMmbed/mbed-os-example-blinky
cd mbed-os-example-blinky
mbed config GCC_ARM_PATH "C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin"
mbed config TOOLCHAIN GCC_ARM
mbed config TARGET LPC1347
mbed compile

----------------------------------------------------------------

JojoS62, 2016-11-17:
I'm using this target also. You can change the targets.json to use the nano lib by adding '"default_lib": "small"' to the LPC1347 section (maybe the key name is 'default_build', that changed recently).
That should work, the side effect is that the nano lib is not thread safe.

I have reported also a problem in #3218, the default stack size is large for this target.

grbd, 2016-11-17:
Thanks for the info I managed to get this working
Example
```
mbed compile -c -m LPC1347 --app-config .\configs\appconfig.json
```

There were two changes I had to make

## Fix 1

The first is to alter the mbed-os\targets\targets.json file
to add “default_lib”: “small”' to the LPC1347 target as you've suggested

```
    "LPC1347": {
        "inherits": ["LPCTarget"],
        "core": "Cortex-M3",
        "extra_labels": ["NXP", "LPC13XX"],
        "supported_toolchains": ["ARM", "GCC_ARM", "IAR"],
        "device_has": ["ANALOGIN", "I2C", "I2CSLAVE", "INTERRUPTIN", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "SERIAL", "SLEEP", "SPI", "SPISLAVE", "STDIO_MESSAGES"],
        "default_lib": "small",
	"release_versions": ["2"],
        "device_name": "LPC1347FBD48"
```

I tried setting this in the application configuration, but because "default_lib" is never set for this target the application config isn't able to override it
Example doesn't work:
```
{
    "target_overrides": {
        "LPC1347": {
	     "inherits": ["LPC1347"],
             "default_lib": "small"
        }
    }
}
```

## Fix 2

The next change is to set the stack size to a smaller value
This is something we can do within the application configuration with
```
{
    "macros": ["__STACK_SIZE=0x100"]
}
```
This gets set within the file mbed-os\targets\TARGET_NXP\TARGET_LPC13XX\device\TOOLCHAIN_GCC_ARM\startup_LPC13xx.S

grbd, 2016-11-17:
This probably would have been easier if we still had custom target support (no need to hack the original mbed library files), but that seems to have been removed recently

JojoS62, 2016-11-17:
the stack settting is more complicated, there is a setting in cmsis_os.h:
```
#    define WORDS_STACK_SIZE   512
```
that turns to
```
#define DEFAULT_STACK_SIZE         (WORDS_STACK_SIZE*4)
```
and this default_stack_size is used in RTX_CM_lib.h:
```
// unless the target is in some exceptions here
static uint32_t thread_stack_main[DEFAULT_STACK_SIZE * 2 / sizeof(uint32_t)];
```
to create a static array for the max stack for the main thread. This you can see in the memory statistics for the rtos/rtx .bss. 
But in my case a bad lib put font data into RAM instead of using 'const' and using the flash memory and caused the overflow. But anyway, a more transparent stack size setting would be fine.

closed 2017-10-27 by ChiefBureaucraticOfficer