ARMmbed/mbed-os #3277: EventQueue unchain + delete = hardfault

tommikas, 2016-11-17:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
GCC_ARM

**meed-os sha:**
(`git log -n1 --oneline`)
f6f872a, aeabcc9

**Expected behavior**
This doesn't cause a hard fault:
```
    EventQueue *eq1 = new EventQueue();
    EventQueue *eq2 = new EventQueue();
    eq2->chain(eq1);
    eq2->chain(NULL);
    delete eq2;
    delete eq1;
```
**Actual behavior**
Hard fault at line: `delete eq2;`
Hard fault doesn't happen if `eq2->chain(NULL);` is removed.

**Steps to reproduce**
Drop the above lines directly into for example mbed-os-example-blinky.

geky, 2016-11-20:
Thanks for reporting this! Looks like the chain function simply didn't check for null like it should have. Should be fixed (with tests) by #3300, let us know if there is still a bug.

tommikas, 2016-11-21:
Nice :) Seems to work fine with #3300.

closed 2016-11-23 by 0xc0170