ARMmbed/mbed-os #3285: Custom Board Support - Enhancement

grbd, 2016-11-17:
### Description
- Type: Enhancement
- Priority: Minor

----------------------------------------------------------------
## Enhancement

I'd like to request some form of custom target / board support.

From what I understand recently this was removed due to implementation issues
But it would be useful to be able to specify a directory as part of the configuration system to look at for custom targets file / source code.
This way additional CPU's / board definitions could be added by "mbed add" as a separate library / module, then used via the config setup.

For myself I'd like to look into the Arduino Due, or custom pcb boards with different pinouts / attached devices.
The current target overrides system doesn't seem to allow for that sort of thing.
Ideally I'd like to avoid using a separate fork of the library / hacking the existing build system

theotherjimmy, 2017-04-25:
@grbd please see #4103

cmonr, 2018-01-23:
Closing due to inactivity. If https://github.com/ARMmbed/mbed-os/pull/4103 is not enough of a solution, feel free to reopen with more details/specifics.

closed 2018-01-23 by cmonr