ARMmbed/mbed-os #3286: mbed export -i IAR -m NCS36510 not working with mbed-os-5.2.2(a1c0840)

radhika-raghavendran, 2016-11-18:
### Description
- Type: Bug 
- Related issue:
- Priority:  Major

---------------------------------------------------------------
## Bug

**Target**
NCS36510

**Toolchain:**
IAR

**Toolchain version:** 7.80.2

**mbed-cli version:** 
0.8.9

**Steps to reproduce**

git checkout a1c0840b3d69060e5eb708edb18358e424a40f51
mbed export -i IAR -m NCS36510

bridadan, 2016-11-18:
Hi @radhika-raghavendran, sorry for the inconvenience and thanks for raising this issue. I just took a look at this and it looks the root cause of this was a check that was added to the IAR exporters to require CMSIS pack support: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/iar/__init__.py#L33.

The CMSIS pack is being used during the export process to automatically set the debugger for the project: https://github.com/ARMmbed/mbed-os/blob/master/tools/export/iar/__init__.py#L116

This has the added benefit of making debugging targets far easier out-of-the-box, but it does now require that targets must have a CMSIS pack.

maclobdell, 2016-11-18:
Our plan going forward is to leverage cmsis packs for exporter information instead of separate data sets.  It seems that we made this change abruptly in 5.2, thus requiring cmsis packs.  We will investigate if we can implement a work around, but the best case would be if you had a preliminary cmsis pack available to test."

maclobdell, 2016-11-18:
oops, accidentally closed and reopened.

radhika-raghavendran, 2016-11-23:
@bridadan @maclobdell 
I thought CMSIS pack was exclusively meant for Kiel MDK IDE. Is there a dependency of IAR on CMSIS pack? Also, is there any information or documentation available on CMSIS pack generation for new devices, which can be used for Kiel MDK, IAR and Eclipse IDEs? There is a link describing CMSIS pack but it addresses only Kiel MDK, http://www.keil.com/pack/doc/CMSIS/Pack/html/createPack_DFP.html

maclobdell, 2016-11-28:
@radhika-raghavendran  I discussed this with @sarahmarshy who is working on the exporters.  cmsis pack isn't really needed for IAR or eclipse IDEs themselves, but the exporter scripts are referencing cmsis packs to set default settings in the generated project (like default debug interface).  This can potentially be done in another way, or maybe if pack is not present, it can just not set a default debugger interface.  We will investigate this and hopefully have a workaround soon.

maclobdell, 2016-11-28:
Sarah just pushed a branch with a patch that could be a workaround to this issue.  It has not been tested yet.  https://github.com/sarahmarshy/mbed/commit/bbb9744cfc33167d698bc870d6b26df0413655c2
If everything works, we can make a pull request to get this on the master branch.

radhika-raghavendran, 2016-11-29:
@maclobdell I tried to bring in the changes on @sarahmarshy local branch into the NCS36510 mbed porting. There is an error during project generation saying that "debugger is not recognized command". If I remove the debugger information in __init__.py, the project generator does through but there is an error in IAR. Can you please have a look at the errors? I have attached the snapshots of the project generator and IAR errors.
[image]
[image]
[image]
[image]
[image]

radhika-raghavendran, 2016-11-29:
@maclobdell @sarahmarshy 
After bringing in the changes in your fork into the mbed-os 5.2.2 release code base, exporting to IAR for NCS36510 is now working. Can you bring this into the main mbed-os repo, so that I can use the main mbed-os branch for development?

radhika-raghavendran, 2016-12-01:
Is the proposed fix merged into the main mbed-os repository? Is there a particular version of mbed-os which I can check?

danclement, 2016-12-01:
@radhika-raghavendran I just imported the mbed blinky example, made sure all my pip tools were updated, and this issue still persists.  This is with the current mbed OS master.

@maclobdell can you give us some guidance on @radhika-raghavendran 's question about version?

theotherjimmy, 2016-12-01:
@radhika-raghavendran @danclement I have not seen any PRs yet, and the file that the above patch modifies remains unmodified in master. I don't think it has been merged yet.

@sarahmarshy The diff looks good to me Could you create a PR for it? (I'll approve the changes)

radhika-raghavendran, 2016-12-02:
@danclement  
The pull request will need to be made on project generator side and only after the merge to main branch will the issue be resolved. 
As mentioned in a previous comment of mine, project generation is working with the changes suggested by @sarahmarshy 
In case you would like to test before the merge to main branch happens, please use the modifications mentioned by @maclobdell and @sarahmarshy

sarahmarshy, 2016-12-02:
Project generator is deprecated. This logic lives in mbed-os tools now. Please see the link to the PR above.

radhika-raghavendran, 2016-12-02:
@sarahmarshy  Can I uninstall project generator and project generator definitions now?
Also, your suggested changes are working and I am able to export the project to IAR. Thank you.

sarahmarshy, 2016-12-02:
Yes, you can!

danclement, 2016-12-02:
Radhika,

I have already gone back to 5.2.2 with the patch from Sarah and I can confirm that I can export an IAR project, open it and compile, and download and debug the NCS36510.  Thank you all for the help.

Best Regards,
Dan Clement

0xc0170, 2016-12-08:
I believe this issue can be closed.

PR #3355 (2016-12-09): IAR export will not fail in the absence of a CMSIS pack
Currently, IAR requires that device has a CMSIS pack to find the debug interface. With that assumption, if a debug interface cannot be found, it defaults to CMSIS-DAP. This change will mean that the default CMSIS-DAP is also used when there is not CMSIS pack at all. 

Fixes #3286

@theotherjimmy
closed 2016-12-08 by 0xc0170