ARMmbed/mbed-os #3292: SingletonPtr<Mutex> mutex; in greentea_metrics.cpp conflicts

Mizuho32, 2016-11-18:
### Description
- Type: Bug(?)
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
Probably all

**Toolchain:**
GCC_ARM

**Toolchain version:**
4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1) 

**mbed-cli version:** 
0.9.10

**meed-os sha:**
a1c0840 Merge pull request #3227 from ARMmbed/release-candidate

This should work.
```cpp
Mutex mutex;
int main(void){
  return 0;
}
```

but in actual, it conflicts with below.
greentea_metrics.cpp
```cpp
/* 37 */ SingletonPtr<Mutex> mutex;
```

static specifier should be added.

adbridge, 2016-11-18:
Hi @Mizuho32 Thanks for finding this and proposing the solution. I believe you are correct . 
There are a couple of points to note however. 
The first is that the release-candidate branch is for our internal use only and should not be used. Please only take a formal release branch or master for your development.
Secondly all PRs must be made against master and not a release branch. Could you please close your PR #3293 and make it against master? Thanks

Mizuho32, 2016-11-18:
Oh, sorry for my ignorance.
I'll rise new one.

closed 2016-11-18 by Mizuho32