ARMmbed/mbed-os #3310: mbed.h (maybe) wrong MBED_PATCH_VERSION

ohagendorf, 2016-11-22:
### Description
- Type: Question maybe Bug 
- Priority: Minor

---------------------------------------------------------------
## Bug

**meed-os sha:**
342e565 Merge pull request #3290 from ARMmbed/release-candidate

in mbed.h the version is defined as 
```
#define MBED_MAJOR_VERSION 5
#define MBED_MINOR_VERSION 2
#define MBED_PATCH_VERSION 1
```

but it is version 5.2.3. Maybe MBED_PATCH_VERSION should be 3 instead of 1.

0xc0170, 2016-11-22:
@ohagendorf Thanks for reporting, will be fixed for the upcoming release. The scripts should be updated to reflect this addition.

cc @adbridge

adbridge, 2016-11-23:
@ohagendorf Thanks for reporting, though please be advised that the release-candidate branch is for internal use only and should not be used for testing. Please only use releases off the mbed-os-5.x branches or master. This particular issue has been fixed for 5.2.3.

closed 2016-11-23 by adbridge