ARMmbed/mbed-os #3314: WiFi broken in mbed OS 5.2.3

jankii01, 2016-11-23:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug | Enhancement | Question
- Related issue: `#abc`
- Priority: Blocker | Major | Minor

---------------------------------------------------------------
## Bug

**Target**
K64F|??

**Toolchain:**
GCC_ARM|ARM|IAR

**Toolchain version:**

**mbed-cli version:** 
(`mbed --version`)

**meed-os sha:**
(`git log -n1 --oneline`)

**DAPLink version:**

**Expected behavior**

**Actual behavior**

**Steps to reproduce**

----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

jankii01, 2016-11-23:
[mbed-os-exmaple-client-logs-wifi-bug.zip](https://github.com/ARMmbed/mbed-os/files/608467/mbed-os-exmaple-client-logs-wifi-bug.zip)

jankii01, 2016-11-23:
It connects;

Connected to Network successfully
IP address 192.168.137.224

However, the TLS handshaking fails.

[DBG ][mClt]: M2MConnectionSecurityPimpl::start_connecting_non_blocking - handshake ret: 1, ssl state: 2

[DBG ][mClt]: receive_handler()

[DBG ][mClt]: receive_handshake_handler()

[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting

[DBG ][mClt]: mbedtls_timing_get_delay

[DBG ][mClt]: mbedtls_timing_get_delay - ret 0

[DBG ][mClt]: receive_from_socket

[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting, ret: -1000

[DBG ][mClt]: ret -1000

[DBG ][mClt]: We wait for next readable event

[DBG ][mClt]: receive_handler()

[DBG ][mClt]: receive_handshake_handler()

[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting

[DBG ][mClt]: mbedtls_timing_get_delay

[DBG ][mClt]: mbedtls_timing_get_delay - ret 0

[DBG ][mClt]: receive_from_socket

[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting, ret: -1000

[DBG ][mClt]: ret -1000

[DBG ][mClt]: We wait for next readable event

[DBG ][mClt]: M2MConnectionSecurityPimpl::timer_expired

[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting

[DBG ][mClt]: mbedtls_timing_get_delay

[DBG ][mClt]: mbedtls_timing_get_delay - ret 1

[DBG ][mClt]: receive_from_socket

[INFO][mClt]: PAL Socket returned: -65522

[DBG ][mClt]: M2MConnectionSecurityPimpl::continue_connecting, ret: -1

[ERR ][mClt]: M2MConnectionSecurityPimpl::timer_expired - handshake timeout

[DBG ][mClt]: handle_connection_error

[DBG ][mClt]: M2MInterfaceImpl::socket_error: (6), retry (1), reconnecting (0)

bulislaw, 2016-11-23:
Wifi example gives errors on K64F as well:
```
pinmap not found for peripheral
```

bulislaw, 2016-11-23:
Ah ignore me, I had some modification for another board. It works fine:
```
Scan:
Network: AccessNG secured: None BSSID: 18:64:72:42:92:0 RSSI: -83 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 18:64:72:42:92:1 RSSI: -82 Ch: 1
Network: AccessNG secured: None BSSID: 9C:1C:12:3:66:20 RSSI: -86 Ch: 1
Network: AccessNG secured: None BSSID: 9C:1C:12:3:b0:80 RSSI: -66 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 9C:1C:12:3:b0:81 RSSI: -66 Ch: 1
Network: iot_test secured: WPA/WPA2 BSSID: 1C:B7:2C:7c:3d:c0 RSSI: -52 Ch: 3
Network: mbed secured: WPA2 BSSID: 50:6A:3:a5:8c:e3 RSSI: -64 Ch: 4
Network: AccessNG secured: None BSSID: 9C:1C:12:3:a9:80 RSSI: -62 Ch: 6
Network: Guest-AccessNG secured: None BSSID: 9C:1C:12:3:a9:81 RSSI: -62 Ch: 6
Network: AccessNG secured: None BSSID: 18:64:72:42:91:e0 RSSI: -82 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 18:64:72:42:91:e1 RSSI: -80 Ch: 1
Network: AccessNG secured: None BSSID: 9C:1C:12:3:6f:80 RSSI: -43 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 9C:1C:12:3:6f:81 RSSI: -42 Ch: 1
Network: AccessNG secured: None BSSID: 9C:1C:12:3:5a:80 RSSI: -78 Ch: 1
Network: Guest-AccessNG secured: None BSSID: 9C:1C:12:3:5a:81 RSSI: -77 Ch: 1
15 networks available.

Connecting...
Success

MAC: 5c:cf:7f:1c:cc:99
IP: 192.168.1.81
Netmask: 255.255.255.0
Gateway: 192.168.1.1
RSSI: -56

Sending HTTP request to www.arm.com...
sent 38 [GET / HTTP/1.1]
recv 64 [HTTP/1.1 200 OK]

Done
```

jankii01, 2016-11-23:
@bulislaw  - the UART WiFi compiled just fine on K64F, I had no issues with pinmap.

bulislaw, 2016-11-23:
Yeah, my bad I was working with another board and forget to change the pins back to default. I assumed that wifi is broken, as title of this issue suggests, it seems that TLS is not connecting properly. I'll update the name so it better reflects nature of the issue.

bulislaw, 2016-11-23:
My bisect points to 1703ed38c902c73b416ce62ccf5d768e801fe2a2 nsapi - Standardized support of NSAPI_UNSPEC

@geky could you have a look please.

adbridge, 2016-11-23:
The associated PR  nsapi - Standardize support of NSAPI_UNSPEC #2897 

It looks like this change was reviewed and tested to some extent by the folks in Oulu, prior to merge. So it looks like this particular case was not considered in the testing.

adbridge, 2016-11-23:
@geky Is this fix that went in for 5.2.3 (https://github.com/ARMmbed/mbed-os/pull/3232) related?

jankii01, 2016-11-23:
Well, @SeppoTakalo  tested it against mesh networking - which is the role the Oulu team has. We do not have WiFi team here. 

Secondly, the change you bisected to this - it has nothing to do with the original issue - that's another change mbedOS team made to the same area (with the same results).

geky, 2016-11-23:
#3232 is unrelated, the DNS query is completing successfully.

Looking at the stack trace of the first socket failure narrowed down the bug pretty quickly.

The esp8266 handles UDP a bit oddly, it actually treats it like a connection based protocol. Because of this, the esp8266 has never returned a valid address from recvfrom. This will likely get fixed in the firmware update that is in progress, but hasn't been high priority since few applications actually use the address from recvfrom.

Client also does not use the resulting address from receive from, but is calling the pal layer in a way that expects an address, causing the application to error even when the data is received successfully. 

The easiest fix is to remove this lines and pass NULL to `pal_receiveFrom` (`pal_receiveFrom`'s documentation indicates this is valid usage):
https://github.com/ARMmbed/mbed-client-classic/blob/master/source/m2mconnectionhandlerpimpl.cpp#L414

---

The reason 1703ed3 reveals the bug is because before pal was interpreting the null address as valid. Adding a check for null (which the pal probably should do) would have also revealed this issue.

And yes, there are tests for this behaviour of recvfrom on there way in, although there are several related issues regarding how these tests are applied generically to network interfaces:
https://github.com/geky/mbed/blob/94ebf499b8bbfe06d7abdfe0b9988143970915ed/features/FEATURE_LWIP/TESTS/mbedmicro-net/udp_echo/main.cpp#L68-L76

geky, 2016-11-23:
Opened a quick pr: https://github.com/ARMmbed/mbed-client-classic/pull/49

adbridge, 2016-11-24:
Closing this as not an mbed-os issue.

closed 2016-11-24 by adbridge

teetak01, 2016-11-25:
Thanks @geky for the quick fix!

jankii01, 2016-11-30:
Turns out "his will likely get fixed in the firmware update that is in progress, but hasn't been high priority since few applications actually use the address from recvfrom." -> well, one of those apps is the provisioning component in Cloud Client. This would really need fixing, as they verify the senders.

kjbracey-arm, 2016-11-30:
An alternative might be to connect your UDP sockets, so you don't need to be checking the source address, but I'm not sure how many of the abstraction layers and network stacks support that. Could be equally glitchy.