ARMmbed/mbed-os #3315: Add support for configurable stack size

mottigondabi, 2016-11-23:
### Description
- Type: Enhancement
- Priority: Major

## Enhancement

**Reason to enhance or problem with existing solution**
There is no way to configure application's main thread stack size at build time.

**Suggested enhancement**
Add `stack_size` attribute to `mbed_app.json` JSON file that will set the initial main thread stack size.

@jenia81

bulislaw, 2016-11-23:
That's sort of configured per target (kind of anyway). We are in process of upgrading RTX to v2, I think that's a good idea, but whether we use default from target and overwrite it if the change is requested in mbed_app.json or assume default main stack size and let people overwrite it in mbed_app.json or maybe we could have one, default configuration that can be overwritten if needed needs to be decided. #3218 #2984

nuket, 2016-11-24:
There were some hints about this here:
https://github.com/ARMmbed/mbed-os/issues/2635#issuecomment-248404271

closed 2017-10-27 by ChiefBureaucraticOfficer