ARMmbed/mbed-os #3325: EFM32GG_STK3700 USB serial is broken at 9600 baud

Patater, 2016-11-25:
### Description
- Type: Bug
- Priority: Blocker

---------------------------------------------------------------
# Bug

**Target**
EFM32GG_STK3700

**Toolchain:**
GCC_ARM

**Toolchain version:**
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

**mbed-cli version:** 
0.9.5

**meed-os sha:**
464f04c6badafe2ce9fd78878b78e6badf300f86

## Expected Behavior
Because the EFM32GG_STK3700 USB serial hardware is capable of 9600 baud, I would expect to be able to use it at 9600 baud with mbed OS.

## Actual Behavior
At 115200 baud, EFM32GG_STK3700 serial works fine. However, if we configure the USB serial port (and our serial client on the PC) for 9600 baud, the PC doesn't receive anything.

## Steps to Reproduce
I've written an example application for the easy reproducing of this issue. By default, it will use 9600 baud. If you change the `baud` variable in the program to 115200, you will receive the following output and some LEDs will blink.
```
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
Hello 115200 World!
```

At 9600 baud, nothing is printed, but the LEDs will blink the same as before.
```C++
// Nothing is printed
```

Here is my example program. (Build with `mbed compile -m EFM32GG_STK3700 -t GCC_ARM` and an mbed-os.lib with `https://github.com/ARMmbed/mbed-os/#464f04c6badafe2ce9fd78878b78e6badf300f86`.)
```C++
// main.cpp
#include "mbed.h"

DigitalOut led1(LED1);
DigitalOut led2(LED2);
Serial pc(USBTX, USBRX);

//int baud = 115200; // This works great!
int baud = 9600; // This doesn't print anything

int main() {
    pc.baud(baud);
    while (1) {
        pc.printf("Hello %d World!\r\n", baud);
        led1 = !led1;
        wait(0.5f);
        led2 = !led2;
        wait(0.5f);
    }
}
```

adbridge, 2016-11-25:
Looks like this https://github.com/ARMmbed/mbed-os/pull/2793 may have been a potential culprit. 
@akselsm Is this board supposed to be able to work at 9600 baud ?

akselsm, 2016-11-25:
The interface firmware is hardcoded to communicate with the EFM32 at 115200 baud, so setting it to anything else on the EFM32 side won't work. It has always been this way (i.e. it's not a regression).

bridadan, 2016-11-28:
I can confirm what @akselsm says, its not a regression.

0xc0170, 2016-11-28:
> The interface firmware is hardcoded to communicate with the EFM32 at 115200 baud, so setting it to anything else on the EFM32 side won't work. It has always been this way (i.e. it's not a regression).

@Patater Does that resolves this? Is the documentation not clear about the baud rate settings? Let us know

Patater, 2016-11-28:
Thanks for the information, guys.

I figured out where my misunderstanding came from. The baud rate I use on the host side (with this particular USB interface) doesn't matter. When my example program is set to use 115200, I can configure my host serial port to 9600 and it still works, printing "Hello 115200 World!". This led me to believe that the board could work at 9600 baud.

So, in practice, the board does work with the host at 9600 baud, but only when the serial port is left with default baud settings (which means the device is using 115200) or explicitly configured for 115200 baud.

closed 2016-11-28 by Patater