ARMmbed/mbed-os #3332: Is there any plan to make Kinetis drivers consistent?

muddog, 2016-11-29:
### Description
- Type: Enhancement
- Priority: Minor

---------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**
The Kinetis targets in the targets/TARGET_Freescale/ folder has two different versions. One is using directly register access for driver APIs, like K20/KLXX. The other is using the Freescale Kinetis SDK in the folder of targets/TARGET_Freescale/TARGET_KSDK2_MCUS/.

**Suggested enhancement**
Could we make them consistent? Using the SDK2.0 would be better.

**Pros**
Using SDK2.0 would be easy to follow Kinetis official release, which brings in immediately bug fix and new features. Also customer would be easy to support other Kinetis or even LPC in the mbedOS.

**Cons**
SDK2.0 would bring in the code size and efficient problem, but looks like a little bit problem.
-----------------------------------------------------------------

0xc0170, 2016-11-29:
cc @mmahadevan108

mmahadevan108, 2016-11-29:
Thanks for your question. 

cc @mnorman4. 

All newer Kinetis targets will be using SDK2.0 drivers. Also all targets under mbed that were using SDK 1.2 have been upgraded to SDK 2.0.

At this point there are no plans to move the non-SDK parts.

0xc0170, 2016-12-07:
I am closing this issue (the answer to the question is that there are no currently any plans for nonSDK parts. As I understand those MCU are not supported in KSDK 2.0. I quickly checked for instance frdm kl25z, its not), the discussion can continue if needed.

closed 2016-12-07 by 0xc0170