ARMmbed/mbed-os #3333: Export to Keil project for new target

ccli8, 2016-11-29:
### Description
- Type: Question
- Priority: Major

## Question
I am porting mbed os 5 to a new target. For debug purpose, I need the function of `Export to Keil project`. As I know, it needed adding info for the new target into `project-generator`/`project-generator-definitions` python packages before. But now, the mechanism seems changed in mbed os 5.2. How do I support this function for my port?

0xc0170, 2016-11-29:
@ccli8 Does this help https://github.com/ARMmbed/mbed-os/blob/master/docs/exporters.md ?

@theotherjimmy @sarahmarshy

ccli8, 2016-11-30:
@0xc0170 According to above link, it seems two stages are needed to support **export to uvision project** for new chip (Nuvoton's M487 in my case):
1. Nuvoton adds M487 into its `.pdsc` file.
1. ARM mbed scans Nuvoton's updated `.pdsc` file to update mbed-os's `index.json`.

It may take some time. Would it be possible for me to add such support for my purpose locally?

theotherjimmy, 2016-11-30:
If it's already there, you could just add the `device_name` field to your targets.json

theotherjimmy, 2016-11-30:
From the doc
> If the device name is not found, we use a generic ARM CPU target in Uvision5.

I don't think you need to have a pack file, so long as the generic ARM CPU target is good enough for your purposes.

Have you tried exporting?

ccli8, 2016-12-01:
@theotherjimmy My modification to `mbed-os` > `targets` > `targets.json` to support Nuvoton's NUMAKER_PFM_M487 board:
```
"NUMAKER_PFM_M487": {
        "core": "Cortex-M4F",
        "default_toolchain": "ARM",
        "extra_labels": ["NUVOTON", "M480", "NUMAKER_PFM_M487"],
        "is_disk_virtual": true,
        "supported_toolchains": ["ARM", "uARM", "GCC_ARM", "IAR"],
        "inherits": ["Target"],
        "device_has": ["I2C", "I2CSLAVE", "I2C_ASYNCH", "INTERRUPTIN", "LOWPOWERTIMER", "PORTIN", "PORTINOUT", "PORTOUT", "PWMOUT", "RTC", "SERIAL", "SERIAL_ASYNCH", "SERIAL_FC", "STDIO_MESSAGES", "SLEEP", "SPI", "SPISLAVE", "SPI_ASYNCH", "TRNG"],
        "release_versions": ["5"],
        "device_name": "M487JIDAE"
    }
```
I try to export to uvision project, but meet with the error:
```
$ mbed export -i uvision -m NUMAKER_PFM_M487
c:\python27\lib\site-packages\fuzzywuzzy\fuzz.py:35: UserWarning: Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning
  warnings.warn('Using slow pure-python SequenceMatcher. Install python-Levenshtein to remove this warning')
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--update-packs]
project.py: error: NUMAKER_PFM_M487 not supported by uvision
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u C:\ccli8\iot\mbed\mbed-os\nu-testtest\mbed-os\tools\project.py -i uvision -m NUMAKER_PFM_M487 --source ." in "C:\ccli8\iot\mbed\mbed-os\nu-testtest"
---
```
This doesn't meet the doc:

> If the device name is not found, we use a generic ARM CPU target in Uvision5.

If I change `device_name` from `M487JIDAE` to e.g. `NUC472HI8AE` (NUMAKER_PFM_NUC472), export is OK. With some fine-tune, I can debug with Keil now.

To support export to uvision project for e.g. NUMAKER_PFM_M487 formally, what is the correct flow in brief?

sarahmarshy, 2016-12-01:
This PR https://github.com/ARMmbed/mbed-os/pull/3349 should fix the issue you are seeing with the generic CPU target.

You have followed the correct flow. Simply add the device name as it is listed in the pdsc, and index.json will need to be generated again to contain the device information. index.json can be updated programmatically. @theotherjimmy can chime in with that process.

0xc0170, 2016-12-07:
@ccli8 Is this issue resolved. The fix referenced above was merged, please test master branch. Reopen if not resolved.

closed 2016-12-07 by 0xc0170

ccli8, 2016-12-07:
@0xc0170 Yes, it is fixed. Thank you.