ARMmbed/mbed-os #3335: Compile USB program with mbed-cli

ccli8, 2016-11-29:
### Description
- Type: Question
- Priority: Minor

---------------------------------------------------------------

## Question
I would like to build USBD program with `mbed-cli` on mbed OS 5.2. For example, to build my main.cpp, I type:
```
mbed compile --source . --source mbed-os --source mbed-os/features/unsupported/USBDevice
```
But I meet the error:
```
[Error] main.cpp@50,0:  #5: cannot open source input file "USBMouse.h": No such file or directory
```
What's the correct command list to build USBD program?

**main.cpp**
```
#include "mbed.h"
#include "USBMouse.h"
 
USBMouse mouse;
 
int main() {
    int16_t x = 0;
    int16_t y = 0;
    int32_t radius = 10;
    int32_t angle = 0;
 
    while (1) {
        x = cos((double)angle*3.14/180.0)*radius;
        y = sin((double)angle*3.14/180.0)*radius;
        
        mouse.move(x, y);
        angle += 3;
        wait(0.001);
    }
}
```

0xc0170, 2016-11-29:
Hello,

usb is not supported for mbed OS 5, therefore it can't be compiled at the moment.

ohagendorf, 2016-11-29:
Is there a plan to reintegrate USB?
Is it only a matter of extending the Python tools to integrate it? Or is it a matter of the driver, API, ... ?

ohagendorf, 2016-11-29:
I asked because I've included USBSerial and USBDevice as a feature (moving from feature/unsupported to feature/usb and extending the tools) for my own projects.

```
mbed compile -t gcc_arm
```
and 

```
mbed export -i gcc_arm
make
```

So after some finetuning a PR would be possible.

bridadan, 2016-11-30:
Hi @ccli8, even though the USBDevice library is unsupported, it should still technically work.

The issue is your call to mbed-cli, you don't need to include the `--source mbed-os` option:

```
mbed compile --source .  --source mbed-os/features/unsupported/USBDevice
```

I tried this locally and it worked just fine, hope that helps!

ccli8, 2016-11-30:
Hi @bridadan, it can work. Many thanks.

closed 2016-11-30 by ccli8

ohagendorf, 2016-11-30:
Hi @bridadan can you say something about the plan to integrate USB as a supported feature?

It is seems to be easy to integrate it but maybe the plan of the mbed team is to change the existing code so that it should stay under unsupported.

bridadan, 2016-12-01:
Hi @ohagendorf, good question! I'd love to see USB come back in as well, though we need to be careful how we do it. The official stance on USB as a supported feature now is that we plan on doing it in the future, but it currently isn't scheduled when this will happen (see @sg-'s response here: https://github.com/ARMmbed/mbed-os/issues/2835#issuecomment-258207911).

I think the USB libraries in general need some cleanup and reorganization, but I haven't looked at them in a while. If you have some thoughts on how this should/could be done, feel free to start a new issue and get a conversation going!