ARMmbed/mbed-os #3337: TRNG implementation for NUMAKER_PFM_NUC472 with GCC throws compilation warnings

andresag01, 2016-11-29:
### Description
- Type: Bug
- Priority: Minor

TRNG implementation for NUMAKER_PFM_NUC472 with GCC throws compilation warnings:
```
[Warning] trng_api.c@89,27: comparison between signed and unsigned integer expressions [-Wsign-compare]
```
---------------------------------------------------------------
## Bug

**Target**
NUMAKER_PFM_NUC472

**Toolchain:**
GCC_ARM

**Toolchain version:**
5.4.1 20160609

**mbed-cli version:** 
0.9.10

**meed-os sha:**
fc1f439 Merge pull request #3323 from andreaslarssonublox/ublox_odin_w2_drivers_update

**Expected behavior**
There should be no warnings

**Actual behavior**
Compiler outputs:
```
Compile [ 97.5%]: trng_api.c
[Warning] trng_api.c@89,27: comparison between signed and unsigned integer expressions [-Wsign-compare]
```
**Steps to reproduce**
mbed compile -m NUMAKER_PFM_NUC472 -t GCC_ARM

0xc0170, 2016-11-29:
@ccli8

cyliangtw, 2016-12-14:
Fix it in PR #3434

cyliangtw, 2016-12-16:
@andresag01 
@0xc0170 
PR#3434 was accepted and merged.
So, please closed it.

bridadan, 2016-12-16:
I'll go ahead and close this per @cyliangtw's recommendation. @andresag01 if you find the issue hasn't been resolved please reopen the issue.

closed 2016-12-16 by bridadan