ARMmbed/mbed-os #3338: 5.3.0-rc1 OOB error: make_armc5 export build error on NUCLEO_F401RE for mbed-os-examples-wifi

bulislaw, 2016-11-29:
### Description
- Type: Bug 
- Priority:  Major

## Bug

**Target**
NUCLEO_F401RE

**Toolchain:**
ARM

**Toolchain version:**
$ armcc -h
Product: ARM Compiler 5.06

**mbed-cli version:** 
$ mbed --version
0.9.1

**meed-os sha:**
5.3.0-rc1

** Example **
 mbed-os-examples-wifi @ mbed-os-5.3.0-rc1

```
$ mbed export -i make_armc5 -m NUCLEO_F401RE
$ make
```
That should produce binary but it fails, while `mbed compile -t arm` works fine.

```
Compile: i2c_api.c
"../mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_ll_usb.c", line 786: Warning:  #3108-D: attribute "__packed__" does not apply here
"../mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_ll_usb.c", line 812: Warning:  #3108-D: attribute "__packed__" does not apply here
../mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_ll_usb.c: 2 warnings, 0 errors
Compile: lp_ticker.c
Compile: mbed_overrides.c
Compile: pinmap.c
Compile: port_api.c
Compile: pwmout_api.c
"../mbed-os/targets/TARGET_STM/TARGET_STM32F4/lp_ticker.c", line 83: Warning:  #1-D: last line of file ends without a newline
../mbed-os/targets/TARGET_STM/TARGET_STM32F4/lp_ticker.c: 1 warning, 0 errors
Compile: rtc_api.c
Compile: serial_api.c
Compile: sleep.c
Compile: spi_api.c
Compile: us_ticker.c
"../mbed-os/targets/TARGET_STM/TARGET_STM32F4/rtc_api_hal.h", line 79: Warning:  #1-D: last line of file ends without a newline
../mbed-os/targets/TARGET_STM/TARGET_STM32F4/rtc_api.c: 1 warning, 0 errors
"../mbed-os/targets/TARGET_STM/TARGET_STM32F4/rtc_api_hal.h", line 79: Warning:  #1-D: last line of file ends without a newline
../mbed-os/targets/TARGET_STM/TARGET_STM32F4/sleep.c: 1 warning, 0 errors
Compile: stm_spi_api.c
Compile: trng_api.c
link: mbed-os-example-wifi.elf
Warning: L6310W: Unable to find ARM libraries.
Error: L6411E: No compatible library exists with a definition of startup symbol __main.
Finished: 0 information, 1 warning and 1 error messages.
make[1]: *** [mbed-os-example-wifi.elf] Error 1
make: *** [all] Error 2
```

bulislaw, 2016-11-29:
@theotherjimmy

geky, 2016-11-29:
Running into this as well
mbed-os-examples-wifi @ mbed-os-5.3.0-rc1
armcc -h => ARM Compiler 5.06 update 2
mbed --version => 0.9.7

theotherjimmy, 2016-11-29:
Yeah, it looks like the compiler can't find its own libraries.

theotherjimmy, 2016-11-29:
I cannot reproduce the issue.

theotherjimmy, 2016-11-29:
Is armcc in your path?

geky, 2016-11-29:
It looks like armcc + other arm tools needs to be in your path for make_armc5 to work, once in path the exported makefile works for me.

0xc0170, 2016-11-30:
Before we resolve this issue, is this a lack of documentation detail, or a user's error?

theotherjimmy, 2016-11-30:
technically, both.

theotherjimmy, 2016-11-30:
https://github.com/ARMmbed/Handbook/blob/master/docs/dev_tools/third_party.md should be updated. I'm going to submit a PR.

theotherjimmy, 2016-11-30:
So, with the PR opened to update the docs, I'm going to close this now.

bulislaw, 2016-12-06:
I'm having the same issue with rc2 on Linux with armcc (from DS-5 package) in my PATH

geky, 2016-12-06:
Just a note, I just found out that the PATH needs to be such that the containing directory of armcc's bin directory is accessible.

ie 
``` bash
ln -s blablabla/ARMCompiler5/arm /arm
export PATH="$PATH:/arm/bin"
```
not
``` bash
ln -s blablabla/ARMCompiler5/arm/bin /armbin
export PATH="$PATH:/armbin"
```

bulislaw, 2016-12-06:
I don't have any links, default ds-5 directory:
`/usr/local/DS-5_v5.24.1/sw/ARMCompiler5.06u2/bin/`

sg-, 2017-02-03:
@bulislaw Is this still an issue?

bulislaw, 2017-02-07:
It works fine now, closing.

closed 2017-02-07 by bulislaw