ARMmbed/mbed-os #3343: `make_iar` exported build fails with unknown include file.

SeppoTakalo, 2016-11-29:
### Description
- Type: Bug

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
IAR

**Steps to reproduce**

```
mbed import <URL of mbed-os-example-mesh-minimal>
mbed export -m K64F -i make_iar
make
```

*Output:* 

```
$ make
C:/Git/mbed-os-example-mesh-minimal/makefile:693: varoitus: syrj▒ytet▒▒n kohteen ".s.o" komennot
C:/Git/mbed-os-example-mesh-minimal/makefile:688: varoitus: ei huomioida vanhoja komentoja kohteelle ".s.o"
Compile: NanostackRfPhyAtmel.cpp

   IAR ANSI C/C++ Compiler V7.70.2.11706/W32 for ARM
   Copyright 1999-2016 IAR Systems AB.
   Network license: 10.118.12.44 (STD)

    #include _DLIB_CONFIG_FILE_HEADER_NAME
                                          ^
"C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.5\arm\inc\c\DLib_Defaults.h",39  Fatal error[Pe1696]: cannot open source file "C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\inc\c\DLib_Config_Full.h"
Fatal error detected, aborting.

Errors: 1
Warnings: none
make[1]: *** [atmel-rf-driver/source/NanostackRfPhyAtmel.o] Virhe 2
make: *** [all] Virhe 2
```
(Sorry about Finnish localisation, ignore those messages)

SeppoTakalo, 2016-11-29:
Is this a problem in my toolchain setup or a valid exporter problem? Can somebody verify?

theotherjimmy, 2016-11-29:
I cannot reproduce the problem

## Versions of things
 * iccarm `IAR ANSI C/C++ Compiler V7.60.2.11341/W32 for ARM`
 * mbed-os-example-mesh-minimal `On branch mbed-os-5.3-rc Your branch is up-to-date with 'origin/mbed-os-5.3-rc'.`
 * mbed-cli `0.9.7`

SeppoTakalo, 2016-11-30:
```
$ iccarm --version
IAR ANSI C/C++ Compiler V7.70.2.11706/W32 for ARM
$ mbed-cli --version
1.0.0
$ cd mbed-os;git describe
mbed-os-5.3.0-rc1
```

Target board used is K64F.

bulislaw, 2016-11-30:
Maybe it's connected to #3338 ?

SeppoTakalo, 2016-11-30:
Actually, I think I found the issue.
mbed-cli seems to hardcode the paths to `C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.3` but I have `7.5`

I need to configure mbed-cli properly.

SeppoTakalo, 2016-11-30:
After setting `IAR_PATH` variable correctly this seems to work. So problem was on my test setup.
(Weird, because I haven't set anything. All others work, except IAR makefile build).

```
C:\Git\mbed-os-example-mesh-minimal>mbed config --list
[mbed] Global config:
ARM_PATH=C:/Keil_v5/ARM
IAR_PATH=C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.5/arm
```

closed 2016-11-30 by SeppoTakalo