ARMmbed/mbed-os #3344: MBED_xxx_VERSION macros should be structured better

adbridge, 2016-11-29:
### Description
- Type: Bug 
- Priority: Minor
---------------------------------------------------------------

In mbed.h we currently have the following structure:
```
#define MBED_LIBRARY_VERSION 130

#if MBED_CONF_RTOS_PRESENT
// RTOS present, this is valid only for mbed OS 5
#define MBED_MAJOR_VERSION 5
#define MBED_MINOR_VERSION 2
#define MBED_PATCH_VERSION 3

#else
// mbed 2
#define MBED_MAJOR_VERSION 2
#define MBED_MINOR_VERSION 0
#define MBED_PATCH_VERSION MBED_LIBRARY_VERSION
#endif
```

There are a number of issues here:
1. We shouldn't be relying on rtos being present to signify that this is an mbed-os 5 release. 
2. For releases we need to automatically update MBED_LIBRARY_VERSION for mbed-os 2 and all 3 of the version macros for mbed-os-5 . With the structure as it stands any code to automatically update them is going to rely on the structure staying the same.

Possible changes:

1. Configure the build system so that when we are building mbed-os 2 we define : 
#define MBED_OS_2
If that isn't defined then the default is mbed-os 5

2. Create a couple of new files: 
a) mbed_versionning_2.h 
   Contains
```
#define MBED_LIBRARY_VERSION 130
#define MBED_MAJOR_VERSION 2
#define MBED_MINOR_VERSION 0
#define MBED_PATCH_VERSION MBED_LIBRARY_VERSION
```

b) mbed_versionning_5.h
   Contains
```
#define MBED_MAJOR_VERSION 5
#define MBED_MINOR_VERSION 2
#define MBED_PATCH_VERSION 3
```

Then in mbed.h we just need something like:
```
#ifdef MBED_OS_2
#include mbed_versionning_2.h
#else
#include mbed_versionning_5.h
#endif
```

That makes the versionning specifically tied to the right build without relying on 3rd party add ons. It also makes the auto updating very easy.

0xc0170, 2016-12-05:
> We shouldn't be relying on rtos being present to signify that this is an mbed-os 5 release.

This is an internal detail, we did not have any other distinction between those 2 code bases. We could introduce  the macro as proposed above, however that would be again for internal use. An app should use mbed version macros. @theotherjimmy what do you think?

> Create a couple of new files

Not fond of having 2 headers for versioning. One might be sufficient ? I understand the reasoning. Is there any other idea ?

cc @c1728p9 @geky @sg-

geky, 2016-12-05:
A simple solution would to just put comments in the source to make parsing easier:
``` cpp
#define MBED_LIBRARY_VERSION 130

#if MBED_CONF_RTOS_PRESENT
// RTOS present, this is valid only for mbed OS 5
#define MBED_MAJOR_VERSION 5 // mbed OS major version
#define MBED_MINOR_VERSION 2 // mbed OS minor version
#define MBED_PATCH_VERSION 3 // mbed OS patch version

#else
// mbed 2
#define MBED_MAJOR_VERSION 2 // mbed 2 major version
#define MBED_MINOR_VERSION 0 // mbed 2 minor version
#define MBED_PATCH_VERSION MBED_LIBRARY_VERSION // mbed 2 patch version
#endif
```

``` bash
sed -i 's|MBED_MINOR_VERSION \d\+ // mbed OS|MBED_MINOR_VERSION 6 // mbed OS|g'
```

--- 

We could also add another macro to make the intention of the rtos test clear:
``` cpp
#define MBED_LIBRARY_VERSION 130

// RTOS present, this is valid only for mbed OS 5
#define MBED_IS_MBED_OS MBED_CONF_RTOS_PRESENT

#if MBED_IS_MBED_OS
#define MBED_MAJOR_VERSION 5 // mbed OS major version
#define MBED_MINOR_VERSION 2 // mbed OS minor version
#define MBED_PATCH_VERSION 3 // mbed OS patch version

#else
// mbed 2
#define MBED_MAJOR_VERSION 2 // mbed 2 major version
#define MBED_MINOR_VERSION 0 // mbed 2 minor version
#define MBED_PATCH_VERSION MBED_LIBRARY_VERSION // mbed 2 patch version
#endif
```

adbridge, 2016-12-06:
@geky Yes good thoughts on how we could improve the parsing of what is there currently, but I was more questionning if this is the best way of doing this in the first place ? Is there not a better way of identifying the software being built ?

bridadan, 2017-01-12:
@adbridge have you explored any other ways of doing this?

adbridge, 2017-01-16:
@bridadan From the discussions had I don't think are any easier ways of improving this other than those already suggested. What we have works for now, if it becomes a problem downline I will look at something based on what @geky suggested. We can probably close this for now.

closed 2017-01-16 by adbridge