ARMmbed/mbed-os #3346: Mesh-minimal build failure due to line endings in PinNames.h

kimlep01, 2016-11-30:
### Description
- Type: Bug
- Priority: Major

Building mbed-os-mesh-minimal against mbed-os master fails in Linux due to file line endings in:
targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/TARGET_UBLOX_EVA_NINA/PinNames.h

[K64F GCC_ARM ATMEL 6lp] + mbed deploy --protocol ssh
[K64F GCC_ARM ATMEL 6lp] error: Your local changes to the following files would be overwritten by checkout:
[K64F GCC_ARM ATMEL 6lp]        targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/TARGET_UBLOX_EVA_NINA/PinNames.h
[K64F GCC_ARM ATMEL 6lp] Please, commit your changes or stash them before you can switch branches.
[K64F GCC_ARM ATMEL 6lp] Aborting
[K64F GCC_ARM ATMEL 6lp] [mbed] ERROR: Unable to update "mbed-os" to rev #d5de476f74dd

Detected originally by CI today (30.11.2016) and can be reproduced manually also.
---------------------------------------------------------------
## Bug

**mbed-cli version:** 
1.0.0

**meed-os sha:**
c57427f (HEAD -> master, origin/master, origin/HEAD) Merge pull request #3321 from jeromecoutant/PR_L432KC

**Steps to reproduce**
Under Linux:
git clone git@github.com:ARMmbed/mbed-os-example-mesh-minimal.git
cd mbed-os-example-mesh-minimal
mbed deploy --protocol ssh
cd mbed-os
git checkout master
git pull
"Cannot pull with rebase: You have unstaged changes."
git diff
"warning: CRLF will be replaced by LF in targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF52832/TARGET_UBLOX_EVA_NINA/PinNames.h."

0xc0170, 2016-11-30:
This is the pull request adding that target and the code referenced above: https://github.com/ARMmbed/mbed-os/pull/3291

cc @mazgch - please have a look

jankii01, 2016-12-01:
Merge this in quickly, please....

0xc0170, 2016-12-01:
> Merge this in quickly, please....

I am looking at this, will fix it

commit bcdb8667 (2016-12-01): ublox eva nina - fix line endings

Fixes #3346
closed 2016-12-01 by 0xc0170