ARMmbed/mbed-os #3351: How to create a simple UDP socket connection in mesh network?

Metaswd, 2016-12-01:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Question
---------------------------------------------------------------
**Target**
Border router: K64F
Router : K64F

**Toolchain:**
GCC_ARM
-----------------------------------------------------------------

## Question

**How to create UDP socket connection in mesh network?**

========== Here is my SENDER(router) socket code ==========
ns_address_t address = {ADDRESS_IPV6, {0xfd, 0x00, 0xdb, 0x80, 0xff, 0x10, 0x00, 0x00,
						0xe8, 0x45, 0x5f, 0xff, 0xfe, 0x72, 0xc6, 0x24}, 8080};  
//this is my border router RF IP address
//(RF bootstrap ready, IPv6 = fd00:db8:ff1::ff:fe00:face)

void socket_callback(void *cb)
{
    socket_callback_t *sock_cb = (socket_callback_t *) cb;
	printf("event type : %d\n", sock_cb->event_type);
	printf("sockid : %d\n", sock_cb->socket_id);
	printf("interface id : %d\n", sock_cb->interface_id);	
	printf("len : %d\n", sock_cb->d_len);
	printf("LQI : %d\n", sock_cb->LINK_LQI);			
}

int main(){
//after mesh.connect success
uint8_t buf[10] = "abcd";
sock_id = socket_open(SOCKET_UDP, 8080, socket_callback);
status = socket_sendto(sock_id, &address, (uint8_t *)buf, sizeof(buf));
}
===============Here is my RECEIVER (Border router) socket code =============
ns_address_t address = {ADDRESS_IPV6, {0xfd, 0x00, 0xdb, 0x80, 0xff, 0x10, 0x00, 0x00,
						0xfe, 0xc2, 0x3d, 0x00, 0x00, 0x04, 0xab, 0x0b}, 8080};
//this is my Router RF IP address
//GP IPv6: fd00:db8:ff1:0:fec2:3d00:4:ab0b

void socket_callback(void *cb)
{
        int8_t status;
        socket_callback_t *sock_cb = (socket_callback_t *) cb;
	printf("event type : %d\n", sock_cb->event_type);
	printf("sockid : %d\n", sock_cb->socket_id);
	printf("interface id : %d\n", sock_cb->interface_id);	
	printf("len : %d\n", sock_cb->d_len);
	printf("LQI : %d\n", sock_cb->LINK_LQI);

	status = socket_read(sock_id, &address, (uint8_t *)buf, sizeof(buf));  
        //i want to read data when i receive data from socket
	printf("read status:%d, buf:%s\n", status, buf);		
}
int main(){
   //after Border Router start success
    sock_id = socket_open(SOCKET_UDP, 8080, socket_callback);
}
=============================================

The result is, the sender will go into callback function,
event type : 144
sockid : 2
interface id : 0
len : 11
LQI : 0

but the receiver has no any response....
so i wonder may i get a simple example of UDP socket sending and receiving in mesh network?

0xc0170, 2016-12-01:
cc @SeppoTakalo

kimlep01, 2016-12-05:
Hi Metaswd,

The socket API works the same way over any network interface. There is a pull request for mbed-os-example-mesh-minimal that shows a simple UDP and TCP example:
https://github.com/ARMmbed/mbed-os-example-mesh-minimal/pull/19

If you have any more questions please let us know.

closed 2017-02-03 by sg-