ARMmbed/mbed-os #3353: LPC15XX CAN still has race condition

odinthenerd, 2016-12-01:
There is a race condition on CANSTAT:

if this
https://github.com/ARMmbed/mbed-os/blob/646cc89a620f978ed6174cb7bbb921e0f0d910f2/targets/TARGET_NXP/TARGET_LPC15XX/can_api.c#L393

and this
https://github.com/ARMmbed/mbed-os/blob/646cc89a620f978ed6174cb7bbb921e0f0d910f2/targets/TARGET_NXP/TARGET_LPC15XX/can_api.c#L455

interrupt each other. 

There is no documentation, to my knowledge, which forbids a can_read or a can_write from being called from ISR context, in fact I am not aware of any documentation at all on this topic. That is in my mind the real problem here.

The possible solution of forbidding both from being called from an ISR is unviable in my mind because it would make it essentially impossible for the user to guarantee that can_read will not be starved to the point that it does not meet its latency grantees. The requirement of provably and reliably meeting latency guarantees is, in my experience, a crucial part of essentially every CAN design. 

- Priority: API design flaw

---------------------------------------------------------------
## Bug

**Target**
LPC15XX

**Toolchain:**
GCC_ARM|ARM|IAR

**Expected behavior**
work

**Actual behavior**
one in a million message loss/timeout

**Suggested enhancement**
redesign the API to include threading and timing guarantees into the public interface. They are part of it any way.

0xc0170, 2016-12-05:
cc @toyowata

devanlai, 2017-02-05:
From section 27.7.4 of [UM10736](http://www.nxp.com/documents/user_manual/UM10736.pdf):
> The value 0x8000 indicates that an interrupt is pending because the CAN Core has updated (not necessarily changed) the Status Register (Error Interrupt or Status Interrupt). 

...

> The CPU has two possibilities to follow the source of a message interrupt:
> • Software can follow the INTID in the Interrupt Register.
> • Software can poll the interrupt pending register, see Section 27.6.3.5.

To me, those two sections imply that:
- Status interrupts will be generated even if the `RXOK` or `TXOK` bits are never reset
- It's preferable to use the per-message-object interrupts for tracking whether an RX or TX (or both) message object is ready.

I did some preliminary testing where I removed all of the instances where the driver was modifying the `CANSTAT` register, which causes the `RXOK` and `TXOK` bits to stay high all the time. As expected, the status interrupt is still generated on every RX/TX event and it calls both the RX and TX IRQ handlers on every status interrupt.

If the user IRQ handlers can handle potentially getting an RX/TX interrupt callback on a TX/RX interrupt, simply never clearing the `RXOK` and `TXOK` bits and always calling both handlers is one solution.

My preferred solution would be to enable the `TXIE`/`RXIE` bits on each message object and then use the `INTID` value or the `INTPND` registers to decide which IRQs to dispatch.

closed 2017-10-27 by ChiefBureaucraticOfficer