ARMmbed/mbed-os #3354: mbedOS 5.3 RC1 - K64F fails to build in Online IDE

jankii01, 2016-12-01:
Note: This is just a template, so feel free to use/remove the unnecessary things

### Description
- Type: Bug
- Related issue: `#abc`
- Priority: Blocker 

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
On-line IDE

**Toolchain version:**
? 

**mbed-cli version:** 
?

**meed-os sha:**
fc1f4391614a88f10f800fd2644e74ed94530f1c

**DAPLink version:**
?
**Expected behavior**

Should  build the binary and start downloading it.

**Actual behavior**
Error -230 it say.

**Steps to reproduce**
Import the mbed-os-example-client as program
delete mbed-os library completely
import it in again as library but by using /#fc1f4391614a88f10f800fd2644e74ed94530f1c as source.
compile all

jankii01, 2016-12-01:
[image]

The actual error note is rather useless, since it's NOT saying WHICH FILE is causing the issue. I'm guess the one KF64 specific .S-file (IAR compiler has sometimes issues that tile, too).

jankii01, 2016-12-01:
[image]

yanesca, 2016-12-01:
I experienced the same with our examples, but then I tried to compile it again and it worked.

0xc0170, 2016-12-01:
cc @theotherjimmy

0xc0170, 2016-12-02:
I imported the example, built and the same error.

@screamerbg - any pointers?

screamerbg, 2016-12-02:
I'm OoO. Cc @theotherjimmy.

adbridge, 2016-12-02:
I tried importing example with 5.2.4 and i got compile error 241

theotherjimmy, 2016-12-02:
I cannot reproduce the error.

My steps:
1) import mbed-os-example-client
2) comment out the error line that @adbridge mentioned
3) click compile button

This produces an executable. Is this the same procedure you guys used?

theotherjimmy, 2016-12-02:
Alright go 2:

Steps:
1) import mbed-os-example-client
2) comment out the line in security.h that @adbridge mentoined
3) delete mbed-os
4) import library from url: https://github.com/ARMmbed/mbed/#fc1f4391614a88f10f800fd2644e74ed94530f1c
5) click compile all

Result: the online IDE downloads an executable.

0xc0170, 2016-12-05:
@jankii01 Can you comment? Does the above helps?

jankii01, 2016-12-05:
Security.h needs to be replaced with a proper security.h that you can get from the mbed.com.

jankii01, 2016-12-05:
Seems to have built now, tried with 5.3 RC2.

closed 2016-12-05 by jankii01