ARMmbed/mbed-os #3357: On-line IDE - allow to take in ANY version of any LIB

jankii01, 2016-12-02:
### Description
- Type: Bug
- Related issue: 
- Priority: Major

---------------------------------------------------------------
## Bug

**Target**
N/A

**Toolchain:**
https://developer.mbed.org/compiler/#

**Toolchain version:**
?

**mbed-cli version:** 
N/A

**meed-os sha:**
N/A

**DAPLink version:**

**Expected behavior**

the on-line IDE should allow me to edit the actual mbed-os.lib for example from blinky via a clear UI (or just show the file and let me edit the hash directly there).
It also should NOT filter out the "non-approved" releases as it also now filters out the future release candidates.

If someone wants to use a specific branch of mbed-OS for any purpose - then that's their issue, the tool should not block it.

**Actual behavior**

You can't see the "unofficial" versions in the revisions list.
You cant' edit the github hash in the .Lib-file

**Steps to reproduce**

Open IDE
import https://github.com/ARMmbed/mbed-os-example-blinky
see mbed-os revision and try to change it to the mbed-OS 5.3 RC1.

Try to see the revision history. 

Nothing's visible from 5.3 branch.
----------------------------------------------------------------
## Enhancement

**Reason to enhance or problem with existing solution**

**Suggested enhancement**

**Pros**

**Cons**

-----------------------------------------------------------------

## Question

**How to?**

0xc0170, 2016-12-02:
cc @sg- @theotherjimmy

sg-, 2017-02-03:
This works when you import from your own fork of mbed-os. The armmbed/mbed-os is a special URL and gives access only to the released versions by design.

closed 2017-02-03 by sg-

JanneKiiskila, 2017-02-06:
I'm not agreeing on this - if someone finds a regression and wants to start bisecting for example - hiding the real "hash" used will just make it worse. Why treat the mbed-os repo any different than any other? I think this a decision that sort of undermines the users.

sg-, 2017-02-07:
Its about setting expectation. Use the right tool for the job.