ARMmbed/mbed-os #3358: K64F ethernet driver enables interrupts before the interrupt handlers are set

kuggenhoffen, 2016-12-02:
### Description
- Type: Bug
- Priority: Minor

---------------------------------------------------------------
## Bug

**Target**
K64F

**Toolchain:**
IAR

**Toolchain version:**
7.80

**mbed-cli version:** 
1.0.0

**meed-os sha:**
a1c0840 (HEAD, tag: mbed_lib_rev129, tag: mbed-os-5.2.2) Merge pull request #3227 from ARMmbed/release-candidate

**DAPLink version:**
0242

**Expected behavior**
Ethernet driver IRQ handlers (`ENET_Transmit_IRQHandler`, `ENET_Receive_IRQHandler` or `ENET_Error_IRQHandler`) should not get called when `s_ENETHandle[0]` is null.

**Actual behavior**
It seems that there is a small time window in `ENET_Init` where interrupts are enabled but the instance handle has not been added to `s_ENETHandle` resulting in null pointer usage in the interrupt handles in case an interrupt happens within this time window.

**Steps to reproduce**
1. Debug mbed-os-example-sockets using IAR
2. Run -> observe that example works correctly
3. Reset (Software reset by default)
4. Run -> observe that hardfault happens

In step 3. if one does hardware reset instead of software, the example works correctly.

0xc0170, 2016-12-02:
cc @geky

0xc0170, 2016-12-08:
cc @mmahadevan108

mmahadevan108, 2016-12-14:
Thank you. Indeed seems like an issue, I have passed this on to the driver developer.

mmahadevan108, 2017-01-12:
Can this be closed?

closed 2017-01-13 by 0xc0170

JanneKiiskila, 2017-01-20:
Should be re-opened. We're still seeing this, unfortunately. Right now with registration-unregistration loop it's popping up with 100% repeatability.

JanneKiiskila, 2017-01-20:
16:11:51.831 | D1 <<< DutThread: [3773][DBG ][mClt]: M2MNsdlInterface::find_resource - token ()
16:11:51.841 | D1 <<< DutThread: assertion "handle != NULL" failed: file "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/fsl_edma.c", line 826, function: EDMA_HandleIRQ

0xc0170, 2017-01-20:
@JanneKiiskila How to reproduce the problem you are seeing?

JanneKiiskila, 2017-01-20:
Repeat mbed-cloud-client registration / unregistration in a loop.
@ilposalo has an automated test case for this.

mmahadevan108, 2017-01-20:
@JanneKiiskila I had tried the mbed-os-example-sockets test that was mentioned when this issue was created. I do not see the hard fault failure with the steps mentioned.

This issue might be a different one, kindly provide the app and steps to reproduce.

JanneKiiskila, 2017-01-20:
Hi @mmahadevan108,

yes indeed it can be a different issue as well. Also this is happening when the device has been running some time already, but I'm afraid I can't share that. Need to see what we can do for a reproducible case.

bridadan, 2017-01-20:
@JanneKiiskila It sounds like there's definitely still an outstanding issue here, though the reproducible case may have changed.

I suggest that we leave this issue closed and you can reopen an updated issue when you have a reproducible case you can share. Does that work for you?

JanneKiiskila, 2017-01-21:
@bridadan - yes, that's the clearest route. We'll try to figure out an isolated test case for this and raise a new issue once we get more clarity on this case.

belkirdi, 2020-01-13:
I have a same issue and I can reproduce based on this code all the time :

#include "EthernetInterface.h"
#include "mbed.h"
#include "rtos.h"
#include "Websocket.h"

#define SLEEP_TIME 500 // (msec)

EthernetInterface eth;
Serial pc(USBTX, USBRX); // tx, rx
bool isEthernetEnabled = false;
bool isEthernetConnecting = false;
bool isWebsocketConnected = false;
const char *ip;

DigitalOut led1(LED1);

void connect_to_internet_using_ethernet() {
  isEthernetConnecting = true;
  isEthernetEnabled = !eth.connect();
  if (isEthernetEnabled) {
    isEthernetConnecting = false;
    printf("Ethernet connected!\r\n");
    ip = eth.get_ip_address();
    const char *gateway = eth.get_gateway();
    const char *netmask = eth.get_netmask();
    const char *mac_address = eth.get_mac_address();
    pc.printf("IP address is: %s\r\n", ip ? ip : "No IP");
    pc.printf("Gateway address is: %s\r\n", gateway ? gateway : "No Gateway");
    pc.printf("Netmask address is: %s\r\n", netmask ? netmask : "No Netmask");
    pc.printf("Mac Address address is: %s\r\n",
           mac_address ? mac_address : "No Mac Address");
    led1 = 0;
  } else {
    led1 = 1;
    pc.printf("Ethernet not able to connect !\r\n");
  }
  isEthernetConnecting = false;
}

void connect_websocket_client_using_ethernet() {
    pc.printf("Connecting websocket client !\r\n");
    Websocket ws((char*) "ws://192.168.1.125:7000/ws", &eth);
    bool is_ws_connected = ws.connect();

    if(is_ws_connected) {
        isWebsocketConnected = true;
        pc.printf("Websocket is connected !\r\n");
    } else {
        isWebsocketConnected = false;
        pc.printf("Websocket is not connected !\r\n");
    }
    
}

void connect_to_internet_using_cellular() {
    // TODO BEK : Not Implemented yet !!
}

void connect_websocket() {
    // TODO BEK : Not Implemented yet !!
}

void disconnect_ws() {
    // TODO BEK : Not Implemented yet !!
}

void reconnect_ws() {
     // TODO BEK : Not Implemented yet !!
}

int main() {
  pc.baud(115200);

  while (true) {  
    ip = eth.get_ip_address();
    if(isEthernetConnecting == false && ip == NULL) {
        connect_to_internet_using_ethernet();
    }

    if(isEthernetEnabled && !isWebsocketConnected) {
        connect_websocket_client_using_ethernet();
    }
    if(isWebsocketConnected) {
        //int res = ws.send((char*) "WebSocket Hello World!");
        //pc.printf("value coming from websocket sending packet (%d)\r\n", res);
    }
    ThisThread::sleep_for(SLEEP_TIME);
    //pc.printf("I am in a loop !!! !\r\n");
  }
}